nvdla/hw #224: Read_reg issue Sanity0 test

shashank-natesh, 2018-09-22:
Hi,
I am facing the following issue while running sanity0.

Compiler version J-2014.12-SP2_Full64; Runtime version J-2014.12-SP2_Full64;  Sep 22 16:21 2018
10570.00ns MSEQ: read_cmd address 0xffff100b with data 0x00000000 and mask 0xffffffe0 (command 4294967292)
1389250.00ns MSEQ: ERROR: Timeout waiting for register read on 0xffff100b, failing.
$finish called from file "/ece/home/nates006/Downloads/hw-nvdlav1/verif/synth_tb/csb_master_seq.v", line 274.
$finish at simulation time    1389750.00ns
           V C S   S i m u l a t i o n   R e p o r t 
Time: 1389750000 ps
CPU Time:      5.830 seconds;       Data structure size: 435.6Mb
Sat Sep 22 16:22:02 2018
Warning: could not find any golden ./*chiplib_dump.raw2
checktest : FAILED : .                                        : all transactions completed : with 1 error. 

Thanks,
Shashank

OopsDylan, 2025-03-12:
HI, do you have any solution to this problem now?

Bhavani001234, 2025-04-09:
Are you able to resolve it.. please help