nvdla/hw #228: about systemc in NVDLA

smartchen2k, 2018-10-08:
hi, as the tutorial told in section "tree.make setup", we should use systemc 2.3.0.
And I found the configuration in gived "tree.make" about systmec is:
`Enter systemc path (Press ENTER if use:
/home/ip/shared/inf/SystemC/1.0/20151112/systemc-2.3.0/GCC472_64_DBG):`
what the GCC472_64_DBG is about?

I doult about this is for that when I run a single test, it always returns a error as:

*Verdi3* WARNING: Failed to load SystemC dumper library `libsscore_vcs201606_sc230_gcc48.so'. Disable SystemC dumping support (/home/CAD/synopsys/Verdi3_L-2016.06-1/share/PLI/lib/LINUXAMD64/libsscore_vcs201606_sc230_gcc48.so: undefined symbol: spi_cast_time_pow_0).

and the simulation test failed. So I think maybe there are some mistakes for my systemc, or the relation between synopsys software and the systemc.

Anybody know about that?

smartchen2k, 2018-10-08:
By the way, verdi works well and the gui can be started.

smartchen2k, 2018-10-08:
And I noly set $PATH for verdi, any other Environment Variables need to set, such as LD_LIBRARY_PATH?

closed 2018-10-10 by smartchen2k

awangsong2012, 2019-03-28:
> hi, as the tutorial told in section "tree.make setup", we should use systemc 2.3.0.
> And I found the configuration in gived "tree.make" about systmec is:
> `Enter systemc path (Press ENTER if use: [...]

hi，Did you solve that problem? I got the same.  HELP Please!

killerzula, 2019-09-09:
Hey @shgangchen did you find a solution? I am facing the same problem.

prasshantg, 2019-09-10:
@killerzula you need to update that path to directory where systemc is installed on your system