nvdla/hw #229: differences between master and nv_small

smartchen2k, 2018-10-11:
in the branch, there are master copy and nv_small copy, and master copy can be configed as nv_small.spec.

So, what's the differences between the master master branch and nv_small branch？

And you said in the document that only nv_small  is tested, whether you mean the nv_small branch or the nv_small configuration?

I'm a little confused.  So @jwise @zdraw @sernc11 @prasshantg @nodushiv , anyone can help to explain that?

mmaciag, 2018-10-11:
So, i was a witness to the last pushes, since I regularly follow the NVDLA development. I am not active NVDLA contributor by the way, except for some little, rather insignificant merge requests.

A long, long time ago there were two branches:
* _nvdlav1_ - oldest branch implementing so called _nv_full_ with floating point support
* _master_ - _nv_small_ fixed configuration development branch

A this point developers worked on _better configurability_. And what they finally did:
* _nvdlav1_ - was left as is (maybe with some minor bugfixes, I didn't follow that)
* master was renamed to _nv_small_
* new master was pushed from their local repositories

The last two points usually is an opportunity to burn someone on the stake by Gitflow fans :)

So to answer your question: in typical git flow, nv_small would be a _tag_ - a stable release of NVDLA supporting nv_small only. And master would just go forth in the tree with newer changes.

Hope it helps :)

smartchen2k, 2018-10-11:
> So, i was a witness to the last pushes, since I regularly follow the NVDLA development. I am not active NVDLA contributor by the way, except for some little, rather insignificant merge requests.
> 
> A long, long time ago there were two branches:
> 
> * _nvdlav1_ - the oldest branch with so called _nv_full_ with floating point support
> * _master_ - development branch of _nv_small_ - although it supports other .spec files it most likely won't synthesise.
> 
> A this point developers worked heavily on _better configurability_. And what they finally did:
> 
> * _nvdlav1_ - was left as is (maybe with some minor bugfixes, I didn't follow that)
> * master was renamed to _nv_small_
> * new master was pushed from their local repositories
> 
> The last two points usually is an opportunity to burn someone on the stake by Gitflow fans :)
> 
> So to answer your question: in typical git flow, nv_small would be a _tag_ - a stable release of NVDLA supporting nv_small only. And master would just go forth in the tree with newer changes.
> 
> Hope it helps :)

Thanks a lot. I really helps.

closed 2018-10-12 by smartchen2k