nvdla/hw #238: how to config fused mode?

xiaoguoer, 2018-10-19:
official document (http://nvdla.org/hw/v1/hwarch.html#functional-description) says "NVDLA has two modes of operation: independent mode and fused mode".
but there's no option about choosing fused mode or not in the .spec file?

mmaciag, 2018-10-22:
I thought this was related to register programming not architecture itself. Each engine can operate independently or can be chained in single pipeline (CC -> SDP -> PDP -> CDP) - architecture should support both (for example see [PDP register](https://github.com/nvdla/hw/blob/nv_small/spec/manual/NVDLA_PDP.rdl) called FLYING_MODE)

xiaoguoer, 2018-10-23:
@mmaciag thank you very much! I understand that now.
But I'm a little curious about what's the function of these .rdl files in the manual directory?
I used "make" command , and some backends(.v, .sv, etc) were generated (as the https://github.com/nvdla/hw/tree/nv_small/spec/manual says). However, I don't know how to use these files......

mmaciag, 2018-10-24:
These 'backends' are used by cmod model (SystemC) and verification suite (SystemVerilog and UVM). I don't think it is used by vmod (actual RTL code). I also think about .rdl files as a supplement to the register documentation: http://nvdla.org/hw/v1/ias/programming_guide.html.

xiaoguoer, 2018-10-24:
@mmaciag It becomes clear now, thank you for replying~~