nvdla/hw #246: How to combine FPGA and nvdla_SW

xiaoguoer, 2018-10-28:
if I intergrated nvdla into FPGA successfully , can I just run "nvdla_runtime" to start  inference with the FPGA conncted to hostPC ? is this feasible?
Or, is nvdla_sw only designed for simulation ??
Than you ~

mmaciag, 2018-10-29:
Running inference on FPGA is perfectly feasible. See this long thread https://github.com/nvdla/hw/issues/110 as an example. So far SoC's were tested (like Zynq UltraScale+) and mixed solutions like Virtex+Zynq. I think there were also attempts to run FPGA with communication over PCIe. I tested Zynq and the software integration is relatively straightforward (as soon as one fixes various minor issues). 

Still, sadly, the main problem is lack of compilation tools for neural networks, that would run on small NVDLA architecture (nv_small or even nv_large). Some teams try to address this issue on their own like iCubeWork: https://github.com/icubecorp/nvdla_compiler - which anyway so far supports nv_full only. Without the compiler you have only little set of sanity tests, you could run with nvdla_runtime.

_nvdla/sw_ of course works also in simulation environment, like QEMU based Virtual Platform (nvdla/vp).

xiaoguoer, 2018-10-30:
@mmaciag thank you for your answer. 
if zynq was used, the steps i thought to run inference are:
1. build kmd/umd according to the linux kernel;
2. copy files which built in the step 1 to OS  running on the zynq;
3. run nvdla_runtime.
am I right ?

mmaciag, 2018-10-30:
Well, in general yes, but maybe not as simple as you described. There are minor things to polish. You definitely want to read about what people were already struggling at nvdla/sw, for example https://github.com/nvdla/sw/issues/46, https://github.com/nvdla/sw/issues/95 and https://github.com/nvdla/sw/issues/70

xiaoguoer, 2018-10-31:
@mmaciag that's definitely what i need, thank you soooo much!
 the [#110 following steps](https://github.com/nvdla/hw/issues/110#issuecomment-377676962) says we should "Add DW02_tree DW_lsb DW_minmax files" and "global define".

the [official doc Designware Components](http://nvdla.org/hw/v1/integration_guide.html#designware-components) says "DW02_tree, DW_lsd and DW_minmax can be obtained directly from the EDA vendors". Does Xilinx provide these components ?

And how to do "global define"?

mmaciag, 2018-10-31:
You should check Vivado (if you work on Zynq) user manual to see how to define global defines. DW_xxx components are not needed when proper macros are defined.

xiaoguoer, 2018-10-31:
@mmaciag okay, thanks a lot ~