nvdla/hw #254: nvdla_small running on FPGA, wrong inference compared to golden results

nookfoo, 2018-12-19:
I am running nvdla_small spec on Xiliinx Ultrascale Board and am able to successfully pass NN_L0_1_small_flatbuf test. Unfortunately the resulting output.dimg has a different md5 hash compared to golden result /o_000000.dimg

This would technically mean inference results are either wrong or deviate. 
Would appreciate if anyone has ideas on what causes this or how to attempt to troubleshoot this case.

Thanks and happy holidays!

shgangchen, 2019-01-15:
@nookfoo hi, did you use the flatbuff under sw?  Did you boot a linux4.3.3 on the ARM and use the KMD and UMD?

nookfoo, 2019-01-15:
@shgangchen Yeah i used flatbuff under sw. I think I used petalinux 2018.2, to boot on xilinx zcu102. I had to cross compile kmd kernel before I could insert it on Petalinux. Don't worry about the questions, I also had no clue in the start. 

I found this thesis project about nvdla on zynq ultrascale where most of the setup steps are described in detail. Maybe it will help: https://webthesis.biblio.polito.it/9033/

shgangchen, 2019-01-16:
@nookfoo thanks for your reply. So there is another question. As far as I know, the software of NVDLA under the sw reposity is currently only supporting the nvdla_full version which is the nvdla1. So how can you run the kmd and umd for the nvdla_small configuration (I think "nvdla_small" you meant is that under "master" branch)? Any modifications? I think this information is quite exciting.

wxbbuaa2011, 2019-01-18:
@nookfoo Thank you very much.

peterzh2018888, 2019-02-25:
@nookfoo @shgangchen @mmaciag 
= Run NN/NN_L0_1_small_fbuf
creating new runtime context...
Emulator starting
submitting tasks...
[ 2210.896668] Enter:dla_read_network_config
[ 2210.903672] Exit:dla_read_network_config status=0
[ 2210.909643] Enter: dla_initiate_processors
[ 2210.915004] Enter: dla_submit_operation
[ 2210.920099] Prepare Convolution operation index 0 ROI 0 dep_count 1
[ 2210.927667] Enter: dla_prepare_operation
[ 2210.932896] processor:Convolution group:1, rdma_group:0 available
[ 2210.940346] Enter: dla_read_config
[ 2210.945125] Exit: dla_read_config
[ 2210.949831] Exit: dla_prepare_operation status=0
[ 2210.955870] Enter: dla_program_operation
[ 2210.961236] Program Convolution operation index 0 ROI 0 Group[1]
[ 2210.968764] no desc get due to index==-1
[ 2210.974179] no desc get due to index==-1
[ 2210.979577] no desc get due to index==-1
[ 2210.984954] no desc get due to index==-1
[ 2210.990313] no desc get due to index==-1
[ 2210.995644] Enter: dla_op_programmed
[ 2211.000630] Update dependency operation index 2 ROI 0 DEP_COUNT=2
[ 2211.008178] Update dependency operation index 1 ROI 0 DEP_COUNT=1
[ 2211.015723] enable SDP in dla_update_dependency as depdency are resolved
[ 2211.023905] Enter: dla_enable_operation
[ 2211.029220] exit dla_enable_operation without actual enable due to processor hasn't been programmed
[ 2211.039839] Exit: dla_enable_operation status=0
[ 2211.045950] Exit: dla_op_programmed
[ 2211.050997] Exit: dla_program_operation status=0
[ 2211.057176] Exit: dla_submit_operation
[ 2211.062484] Enter: dla_dequeue_operation
[ 2211.067953] Dequeue op from Convolution processor, index=2 ROI=0
[ 2211.075536] Enter: dla_submit_operation
[ 2211.080938] Prepare Convolution operation index 2 ROI 0 dep_count 1
[ 2211.088792] Enter: dla_prepare_operation
[ 2211.094292] processor:Convolution group:0, rdma_group:0 available
[ 2211.101991] Enter: dla_read_config
[ 2211.106995] Exit: dla_read_config
[ 2211.111900] Exit: dla_prepare_operation status=0
[ 2211.118105] Enter: dla_program_operation
[ 2211.123600] Program Convolution operation index 2 ROI 0 Group[0]
[ 2211.131226] no desc get due to index==-1
[ 2211.136737] no desc get due to index==-1
[ 2211.142218] no desc get due to index==-1
[ 2211.147668] no desc get due to index==-1
[ 2211.153083] no desc get due to index==-1
[ 2211.158471] Enter: dla_op_programmed
[ 2211.163499] Update dependency operation index 6 ROI 0 DEP_COUNT=3
[ 2211.171072] Update dependency operation index 3 ROI 0 DEP_COUNT=2
[ 2211.178616] Exit: dla_op_programmed
[ 2211.183552] Exit: dla_program_operation status=0
[ 2211.189625] Exit: dla_submit_operation
[ 2211.194807] Exit: dla_dequeue_operation
[ 2211.200064] Enter: dla_submit_operation
[ 2211.205327] Prepare SDP operation index 1 ROI 0 dep_count 0
[ 2211.212329] Enter: dla_prepare_operation
[ 2211.217670] processor:SDP group:0, rdma_group:1 available
[ 2211.224484] Enter: dla_read_config
[ 2211.229290] Exit: dla_read_config
[ 2211.233999] Exit: dla_prepare_operation status=0
[ 2211.240023] Enter: dla_program_operation
[ 2211.245346] Program SDP operation index 1 ROI 0 Group[0]
[ 2211.252076] no desc get due to index==-1
[ 2211.257384] no desc get due to index==-1
[ 2211.262694] no desc get due to index==-1
[ 2211.268009] no desc get due to index==-1
[ 2211.273323] Enter: dla_op_programmed
[ 2211.278305] Update dependency operation index 3 ROI 0 DEP_COUNT=1
[... 133 lines cut ...]
[ 2212.042876] Exit:dla_handle_events, ret:0
[ 2212.048188] Enter:dla_handle_events, processor:SDP
[ 2212.054317] Exit:dla_handle_events, ret:0
[ 2212.059688] Enter:dla_handle_events, processor:PDP
[ 2212.065855] Exit:dla_handle_events, ret:0
[ 2212.071217] Enter:dla_handle_events, processor:CDP
[ 2212.077357] Exit:dla_handle_events, ret:0
[ 2212.082717] Enter:dla_handle_events, processor:RUBIK
[ 2212.089054] Exit:dla_handle_events, ret:0
[ 2212.094454] Enter:dla_handle_events, processor:BDMA
[ 2212.100733] Exit:dla_handle_events, ret:0
[ 2212.106130] Enter:dla_handle_events, processor:Convolution
[ 2212.113015] Exit:dla_handle_events, ret:0
[ 2212.118409] Enter:dla_handle_events, processor:SDP
[ 2212.124584] Exit:dla_handle_events, ret:0
[ 2212.129985] Enter:dla_handle_events, processor:PDP
[ 2212.136181] Exit:dla_handle_events, ret:0
[ 2212.141594] Enter:dla_handle_events, processor:CDP
[ 2212.147788] Exit:dla_handle_events, ret:0
[ 2212.153183] Enter:dla_handle_events, processor:RUBIK
[ 2212.159561] Exit:dla_handle_events, ret:0

Dead here, can't keep running.

What could be the cause of this problem? Has anyone ever had the same problem? How to fix it.

Okaymaddy, 2019-09-18:
@nookfoo @shgangchen What are the files you are adding for successful synthesis and implementation? I am using: 
- /outdir/nv_small/vmod/nvdla
- /outdir/nv_small/vmod/include
- /vmod/fifos
- /vmod/rams/fpga/
- outdir/nv_small/vmod/vlibs

Okaymaddy, 2019-10-29:
> Hi @shgangchen, the current KMD and UMD actually does have support(*) for nv_small - what is missing for public is the compiler that would generate those flatbufs (like NN_L0_1_small_flatbuf). The available flatbuffs come from here: https://github.com/nvdla/sw/tree/master/regression/flatbufs/kmd
> 
> From other threads I learned that compiler for nv_small should be released by end of Q1.
> 
> (*) as for KMD - compile with `-DDLA_SMALL_CONFIG`

@mmaciag I looked at `nvdla/sw/kmd/firmware/include/opendla.h` and the flag I see there is: `DLA_2_CONFIG` to access `opendla_small.h`. I am not sure if they changed the flag name.

[image]

ccc120660, 2023-12-04:
> @nookfoo @shgangchen @mmaciag = Run NN/NN_L0_1_small_fbuf creating new runtime context... Emulator starting submitting tasks... [ 2210.896668] Enter:dla_read_network_config [ 2210.903672] [...]

Same issue,is there a solution？