nvdla/hw #265: Can nvdla_small spec be modified with secondary SRAMIF and batch mode for NN_L0_1_small_fbuf?

nookfoo, 2019-02-09:
Hello there,

I am running nvdla_small on ZCU102 FPGA and am curioues to find out if enabling batch mode or the secondary memory interface sramif will improve performance. I intend to try it out anyway, but wanted to ask beforehand, if this is even sensible since NN_L0_1_small_fbuf cannot be modified due to unreleased compiler.

Can NN_L0_1_small_fbuf utilize batch mode or secondary sramif enabled in nvdla_small?

nookfoo, 2019-02-11:
Update:

After enabling Batch mode and setting MAX_BATCH_SIZE to 32 (an different value is not accepted by tmake) for nvdla_small spec and nvdla_medium_512 spec, Synthesis of rtl code for both specs is failing.
Synthesis error revolves around 'u_lat_fifo' instance of 'NV_NVDLA_SDP_MRDMA_EG_lat_fifo' module having a non-existant port connection 'nvdla_core_clk'. 
I had clock gating and fpga macros enabled and made the same implementation steps as for my running nvdla_small.

Either batch mode is not possible for these specs or the code has not been properly tested for these configurations. Will be testing secondary SRAMIF next.

JSnobody, 2019-02-12:
@nookfoo Have you ever run nvdla_small on ZCU102 FPGA successfully？Maybe we can talk together.

mmaciag, 2019-02-13:
@nookfoo Let me know about your findings about batch mode. From my previous observations enabling it hardly increases the resource usage in FPGA, which was quite confusing.

Enabling `sramif` would require `BDMA` which is kind of separate engine. My guess is that `NN_L0_1_small_fbuf` would need to _physically_ contain instructions (operations) handling `BDMA`.

I don't know if _sramif_ will work with current _nv_small_ spec, but at least you can measure the actual DDR performance with Xilinx' AXI Performance Monitor IP core. Nice thing about this core is that it measures also peak wait state cycles on the AXI4 bus.

With DDR4 2400, 64-bit on the PS side it is hardly possible to use entire bandwidth from PL side (with single AXI4 bus). However you may observe quite decent wait cycles. For example FPGA running at 250 MHz, one channel write (AXI4 Traffic Generator IP Core), gave us following performance:
- Average bandwidth: 3.711 GB/s (95% of theoretical bandwidth for 128-bit AXI4 bus)
- Peak latency (wait state): 1507 cycles - I guess this is where you would like to use `PRIMARY_MEMIF_LATENCY_1024` in place of `PRIMARY_MEMIF_LATENCY_64`.

Alas ZU9EG does not have URAM block, which would be very handful for efficient `sramif` implementation.

peterzh2018888, 2019-02-24:
root@aidla:/# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko
[ 170.618471] reset engine done
[ 170.621785] [drm] Initialized nvdla 0.0.0 20171017 for a0000000.nvdla_small on minor 1
[ 170.629694] NVDLA a0000000.nvdla_small: failed to register drm device
[ 170.636085] NVDLA: probe of a0000000.nvdla_small failed with error -12
root@aidla:/# ls /dev/dri/
card0 card1 renderD128
root@aidla:/# cat /proc/interrupts
CPU0 CPU1 CPU2 CPU3
3: 2520 4132 10902 1416 GICv2 30 Level arch_timer
6: 0 0 0 0 GICv2 67 Level ff9905c0.mailbox
7: 0 0 0 0 GICv2 175 Level arm-pmu
8: 0 0 0 0 GICv2 176 Level arm-pmu
9: 0 0 0 0 GICv2 177 Level arm-pmu
10: 0 0 0 0 GICv2 178 Level arm-pmu
13: 0 0 0 0 GICv2 156 Level zynqmp-dma
14: 0 0 0 0 GICv2 157 Level zynqmp-dma
15: 0 0 0 0 GICv2 158 Level zynqmp-dma
16: 0 0 0 0 GICv2 159 Level zynqmp-dma
17: 0 0 0 0 GICv2 160 Level zynqmp-dma
18: 0 0 0 0 GICv2 161 Level zynqmp-dma
19: 0 0 0 0 GICv2 162 Level zynqmp-dma
20: 0 0 0 0 GICv2 163 Level zynqmp-dma
21: 0 0 0 0 GICv2 164 Level Mali_GP_MMU, Mali_GP, Mali_PP0_MMU, Mali_PP0, Mali_PP1_MMU, Mali_PP1
22: 0 0 0 0 GICv2 109 Level zynqmp-dma
23: 0 0 0 0 GICv2 110 Level zynqmp-dma
24: 0 0 0 0 GICv2 111 Level zynqmp-dma
25: 0 0 0 0 GICv2 112 Level zynqmp-dma
26: 0 0 0 0 GICv2 113 Level zynqmp-dma
27: 0 0 0 0 GICv2 114 Level zynqmp-dma
28: 0 0 0 0 GICv2 115 Level zynqmp-dma
29: 0 0 0 0 GICv2 116 Level zynqmp-dma
31: 0 0 0 0 GICv2 95 Level eth0, eth0
33: 484 0 0 0 GICv2 49 Level cdns-i2c
34: 43 0 0 0 GICv2 50 Level cdns-i2c
35: 0 0 0 0 GICv2 42 Level ff960000.memory-controller
36: 0 0 0 0 GICv2 57 Level axi-pmon
37: 0 0 0 0 GICv2 150 Level nwl_pcie:misc
42: 23 0 0 0 GICv2 47 Level ff0f0000.spi
43: 0 0 0 0 GICv2 58 Level ffa60000.rtc
44: 0 0 0 0 GICv2 59 Level ffa60000.rtc
45: 0 0 0 0 GICv2 165 Level ahci-ceva[fd0c0000.ahci]
46: 1749 0 0 0 GICv2 81 Level mmc0
47: 365 0 0 0 GICv2 53 Level xuartps
49: 0 0 0 0 GICv2 145 Edge fd4d0000.watchdog
50: 0 0 0 0 GICv2 88 Level ams-irq
51: 0 0 0 0 GICv2 154 Level fd4c0000.dma
52: 0 0 0 0 GICv2 151 Level fd4a0000.zynqmp-display
62: 0 0 0 0 GICv2 97 Level xhci-hcd:usb1
65: 0 0 0 0 zynq-gpio 22 Edge sw19
IPI0: 2197 1885 1916 1401 Rescheduling interrupts
IPI1: 7 3 8 8 Function call interrupts
IPI2: 0 0 0 0 CPU stop interrupts
IPI3: 0 0 0 0 CPU stop (for crash dump) interrupts
IPI4: 2192 1898 955 2321 Timer broadcast interrupts
IPI5: 0 0 0 0 IRQ work interrupts
IPI6: 0 0 0 0 CPU wake-up interrupts
Err: 0
root@aidla:/# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko
insmod: can't insert '/lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko': File exists
root@aidla:/#
@mmaciag NVDLA a0000000.nvdla_small: failed to register drm device。cat /proc/interrupts not right。Why?

peterzh2018888, 2019-02-24:
root@aidla:/# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko
[ 170.618471] reset engine done
[ 170.621785] [drm] Initialized nvdla 0.0.0 20171017 for a0000000.nvdla_small on minor 1
[ 170.629694] NVDLA a0000000.nvdla_small: failed to register drm device
[ 170.636085] NVDLA: probe of a0000000.nvdla_small failed with error -12
root@aidla:/# ls /dev/dri/
card0 card1 renderD128
root@aidla:/# cat /proc/interrupts
CPU0 CPU1 CPU2 CPU3
3: 2520 4132 10902 1416 GICv2 30 Level arch_timer
6: 0 0 0 0 GICv2 67 Level ff9905c0.mailbox
7: 0 0 0 0 GICv2 175 Level arm-pmu
8: 0 0 0 0 GICv2 176 Level arm-pmu
9: 0 0 0 0 GICv2 177 Level arm-pmu
10: 0 0 0 0 GICv2 178 Level arm-pmu
13: 0 0 0 0 GICv2 156 Level zynqmp-dma
14: 0 0 0 0 GICv2 157 Level zynqmp-dma
15: 0 0 0 0 GICv2 158 Level zynqmp-dma
16: 0 0 0 0 GICv2 159 Level zynqmp-dma
17: 0 0 0 0 GICv2 160 Level zynqmp-dma
18: 0 0 0 0 GICv2 161 Level zynqmp-dma
19: 0 0 0 0 GICv2 162 Level zynqmp-dma
20: 0 0 0 0 GICv2 163 Level zynqmp-dma
21: 0 0 0 0 GICv2 164 Level Mali_GP_MMU, Mali_GP, Mali_PP0_MMU, Mali_PP0, Mali_PP1_MMU, Mali_PP1
22: 0 0 0 0 GICv2 109 Level zynqmp-dma
23: 0 0 0 0 GICv2 110 Level zynqmp-dma
24: 0 0 0 0 GICv2 111 Level zynqmp-dma
25: 0 0 0 0 GICv2 112 Level zynqmp-dma
26: 0 0 0 0 GICv2 113 Level zynqmp-dma
27: 0 0 0 0 GICv2 114 Level zynqmp-dma
28: 0 0 0 0 GICv2 115 Level zynqmp-dma
29: 0 0 0 0 GICv2 116 Level zynqmp-dma
31: 0 0 0 0 GICv2 95 Level eth0, eth0
33: 484 0 0 0 GICv2 49 Level cdns-i2c
34: 43 0 0 0 GICv2 50 Level cdns-i2c
35: 0 0 0 0 GICv2 42 Level ff960000.memory-controller
36: 0 0 0 0 GICv2 57 Level axi-pmon
37: 0 0 0 0 GICv2 150 Level nwl_pcie:misc
42: 23 0 0 0 GICv2 47 Level ff0f0000.spi
43: 0 0 0 0 GICv2 58 Level ffa60000.rtc
44: 0 0 0 0 GICv2 59 Level ffa60000.rtc
45: 0 0 0 0 GICv2 165 Level ahci-ceva[fd0c0000.ahci]
46: 1749 0 0 0 GICv2 81 Level mmc0
47: 365 0 0 0 GICv2 53 Level xuartps
49: 0 0 0 0 GICv2 145 Edge fd4d0000.watchdog
50: 0 0 0 0 GICv2 88 Level ams-irq
51: 0 0 0 0 GICv2 154 Level fd4c0000.dma
52: 0 0 0 0 GICv2 151 Level fd4a0000.zynqmp-display
62: 0 0 0 0 GICv2 97 Level xhci-hcd:usb1
65: 0 0 0 0 zynq-gpio 22 Edge sw19
IPI0: 2197 1885 1916 1401 Rescheduling interrupts
IPI1: 7 3 8 8 Function call interrupts
IPI2: 0 0 0 0 CPU stop interrupts
IPI3: 0 0 0 0 CPU stop (for crash dump) interrupts
IPI4: 2192 1898 955 2321 Timer broadcast interrupts
IPI5: 0 0 0 0 IRQ work interrupts
IPI6: 0 0 0 0 CPU wake-up interrupts
Err: 0
root@aidla:/# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko
insmod: can't insert '/lib/modules/4.14.0-xilinx-v2018.2/extra/opendla.ko': File exists
root@aidla:/#
@nookfoo  NVDLA a0000000.nvdla_small: failed to register drm device。cat /proc/interrupts not right。Why?

peterzh2018888, 2019-02-24:
@nookfoo @JSnobody Have you ever run nvdla_small on ZCU102 FPGA successfully？Maybe we can talk together.
mine is NVDLA a0000000.nvdla_small: failed to register drm device。cat /proc/interrupts not right。Why?

peterzh2018888, 2019-02-24:
> Hello there,
> 
> I am running nvdla_small on ZCU102 FPGA and am curioues to find out if enabling batch mode or the secondary memory interface sramif will improve performance. I intend to try it out anyway, but wanted to ask beforehand, if this is even sensible since NN_L0_1_small_fbuf cannot be modified due to unreleased compiler.
> 
> Can NN_L0_1_small_fbuf utilize batch mode or secondary sramif enabled in nvdla_small spec?

I am running nvdla_small spec on Xiliinx Ultrascale Board ZCU102 But: NVDLA a0000000.nvdla_small: failed to register drm device。cat /proc/interrupts not right。There's a problem ,Why,how to solve it? Did you have the same problem?

peterzh2018888, 2019-02-25:
> run nvdla_small on ZCU102 FPGA successfully

I had run nvdla_small on ZCU102 FPGA successfully.But There's a problem ,Why,how to solve it? Did you have the same problem like this?

Run NN/NN_L0_1_small_fbuf
creating new runtime context...
Emulator starting
submitting tasks...
Run NN/NN_L0_1_small_fbuf can't pass ,but still submitting tasks.. Have you ever had the same problem?

peterzh2018888, 2019-02-25:
> @nookfoo Let me know about your findings about batch mode. From my previous observations enabling it hardly increases the resource usage in FPGA, which was quite confusing.
>
> Enabling `sramif` would [...]
I had run nvdla_small on ZCU102 FPGA successfully.But There's a problem ,Why,how to solve it? Did you have the same problem like this?

Run NN/NN_L0_1_small_fbuf
creating new runtime context...
Emulator starting
submitting tasks...
Run NN/NN_L0_1_small_fbuf can't pass ,but still submitting tasks.. Have you ever had the same problem?

mmaciag, 2019-02-25:
@peterzh2018888 Have you _ever_ heard about "_One bug, one bug report?_" rule? No matter if it's a bug, or problem in your setup - in last two days I've got roughly 17 notifications from you describing the same or similar problem. It starts to be indistinguishable from spamming...

Don't understand me wrong... People _are_ reading your posts. For example I am subscribed to both `sw` and `hw` projects, and I see _everything_ what's happening here. But this is not a big community, so it is unlikely to get response immediately (and hopefully the maintaners are busy with releasing the compiler :+1:). 

You may increase your chances by describing the problem in _detail_, including the environment setup (compiler version, compiler flags, ...) and steps to reproduce the problem. Few people already worked on Zynq. And as I recall some already had problems with DRM and interrupts so maybe they will be kind enough to compare your workflow with theirs... By spamming them you only increase the level of their annoyance :)

peterzh2018888, 2019-02-25:
You are right. People are reading my posts. But did't  feed back to me. I just need rather hurryy feed back  for me even it is unlikely to get response immediately.I will increase my chances by describing the problem in detail, including the environment setup (compiler version, compiler flags, ...) and steps to reproduce the problem.Thank you.

peterzh2018888, 2019-02-25:
@mmaciag You are right. People are reading my posts. But did't  feed back to me. I just need rather hurryy feed back  for me even it is unlikely to get response immediately.I will increase my chances by describing the problem in detail, including the environment setup (compiler version, compiler flags, ...) and steps to reproduce the problem.Thank you.

shgangchen, 2019-02-28:
hi @nookfoo , I received your update about the sythesis problem, however, I can't find it here. Is that not a problem anymore?

huangwei858, 2019-03-11:
> @peterzh2018888 Have you _ever_ heard about "_One bug, one bug report?_" rule? No matter if it's a bug, or problem in your setup - in last two days I've got roughly 17 notifications from you [...]

Hey, I've send some questions in another issues about how to config BRAM replace RAMDP/RAMPDP, could you share your experience in logical ram wrapper instead of simulation ram