nvdla/hw #270: nv small with *_DATA_TYPE_INT16_FP16 Error

nsherer, 2019-03-14:
Hi,

Trying to generate an nv_small design with both:
#define FEATURE_DATA_TYPE_INT16_FP16
#define WEIGHT_DATA_TYPE_INT16_FP16

but receiving the below error:
------------------------------------------------------
ERROR: eperl: eval:
my $i;
for($i = 0; $i < 32; $i ++) {
$b0 = 16 * $i;
$b1 = 16 * $i + 15;
print "assign wt_nan_mask[${b1}:${b0}] = {16{~wt_fp16_nan_flag_w[${i}]}};\n";
}
print "\n\n";

error:
Global symbol "$b0" requires explicit package name at (eval 41) line 3, <IN1> line 994.
Global symbol "$b1" requires explicit package name at (eval 41) line 4, <IN1> line 994.
Global symbol "$b1" requires explicit package name at (eval 41) line 5, <IN1> line 994.
Global symbol "$b0" requires explicit package name at (eval 41) line 5, <IN1> line 994.

 at ../../../tools/bin/eperl line 265
	eperl_main::Error('eval:\x{a}my $i;\x{a}for($i = 0; $i < 32; $i ++) {\x{a}$b0 = 16 * $i;\x{a}$b1...') called at ../../../tools/bin/eperl line 210
	eperl_main::EvalScript('my $i;\x{a}for($i = 0; $i < 32; $i ++) {\x{a}$b0 = 16 * $i;\x{a}$b1 = 16 ...') called at ../../../tools/bin/eperl line 189
	eperl_main::ProcessInput('../../../outdir/nv_small/vmod/nvdla/cdma/NV_NVDLA_CDMA_wt.v.vcp', '../../../outdir/nv_small/vmod/nvdla/cdma/NV_NVDLA_CDMA_wt.v.v...') called at ../../../tools/bin/eperl line 158
make: *** [../../../outdir/nv_small/vmod/nvdla/cdma/NV_NVDLA_CDMA_wt.v] Error 25
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE
------------------------------------------------------

Why is that?

huangwei858, 2019-03-14:
You have to check eperl script settings about your error. nv_small should be set as correct settings or modify reference perl script base on you requirements.

nsherer, 2019-03-14:
Can you please provide such script.
we would like to have the below confs,
please try to compile it and let me know where am I wrong,
Should be very simple - only enlarged the width of the DATA/WT to 16 bits

Thx

our spec/defs/nv_small.spec is as below:

//#define NV_SMALL 1
#define FEATURE_DATA_TYPE_INT16_FP16
#define WEIGHT_DATA_TYPE_INT16_FP16
#define WEIGHT_COMPRESSION_DISABLE
#define WINOGRAD_DISABLE 
#define BATCH_DISABLE
#define SECONDARY_MEMIF_DISABLE
#define SDP_LUT_DISABLE
#define SDP_BS_ENABLE
#define SDP_BN_ENABLE
#define SDP_EW_DISABLE
#define BDMA_DISABLE
#define RUBIK_DISABLE
#define RUBIK_CONTRACT_DISABLE
#define RUBIK_RESHAPE_DISABLE
#define PDP_ENABLE
#define CDP_ENABLE
#define RETIMING_DISABLE
#define MAC_ATOMIC_C_SIZE_8
#define MAC_ATOMIC_K_SIZE_8
#define MEMORY_ATOMIC_SIZE_8
#define MAX_BATCH_SIZE_x
#define CBUF_BANK_NUMBER_32
#define CBUF_BANK_WIDTH_8
#define CBUF_BANK_DEPTH_512
#define SDP_BS_THROUGHPUT_1
#define SDP_BN_THROUGHPUT_1
#define SDP_EW_THROUGHPUT_x
#define PDP_THROUGHPUT_1
#define CDP_THROUGHPUT_1
#define PRIMARY_MEMIF_LATENCY_50
#define SECONDARY_MEMIF_LATENCY_x
//#define PRIMARY_MEMIF_MAX_BURST_LENGTH_4
#define PRIMARY_MEMIF_MAX_BURST_LENGTH_1
#define PRIMARY_MEMIF_WIDTH_64
#define SECONDARY_MEMIF_MAX_BURST_LENGTH_x
#define SECONDARY_MEMIF_WIDTH_512
#define MEM_ADDRESS_WIDTH_32
#define NUM_DMA_READ_CLIENTS_7
#define NUM_DMA_WRITE_CLIENTS_3

#include "projects.spec"

killerzula, 2019-09-19:
Doesn't nv_small only support INT8 configuration?

nsherer, 2019-09-19:
Hi,

We overcame that issue. modified the small to support also INT16.

Thanks anyway,
Neta

Neta Sherer

M: +972 54 7916089 <%2B972%2054%202522023> | T: +972 77 2518840
<%2B972%2077%202518844>

E: neta.sherer@adasky.com | *www.adasky.com <http://www.adasky.com/>*

-- 
_CONFIDENTIALITY NOTICE  This email message and any attachments thereto are 
confidential and/or privileged and/or subject to privacy laws and are 
intended only for use by the addressee(s) named above. If you are not the 
intended addressee or reader of this email, you are hereby notified that 
any dissemination, distribution, copying or use of this email and any 
attachments thereto is strictly prohibited. If you have received this 
message in error, please notify the sender by reply email or by telephone 
and delete or destroy any copy of this message (and any attachments 
included thereto). _
_
_

killerzula, 2019-09-19:
How, if I may ask?

nsherer, 2020-08-30:
Just used the supplied scripts and modified it accordingly.

killerzula, 2020-08-30:
Bro, I'm working on a different project now. Thanks for replying though.

LeiWang1999, 2021-03-17:
Hi, @nsherer
What the branch did you use ?
I notice that the master branch project.spec can only support int8 feature/weight datatype rtl generate.
But the nvdlav1 branch can not generate fpga ram dir.