nvdla/hw #274: Compilation/synthesis issues in Vivado for the nv_small config in both nv_small and master branches

davide-giri, 2019-03-22:
Hello,

Can anybody point me to a stable branch and commit for what concerns the RTL of the nv_small configuration? Please let me know if you were able to compile and synthesize the RTL sources in Vivado without errors.

I'm working on deploying the nv_small on FPGA. The current status of both the `nv_small` and the `master` branches doesn't seem to be stable. I built a block design in Vivado for both branches and I incur multiple compilation and synthesis errors. Most of them are minor, but I'm afraid to be working with an unstable design that will be functionally incorrect once deployed on FPGA.

Thanks in advance!

mmaciag, 2019-03-22:
@davide-giri Hi, I can confirm that both *master* and *nv_branch* can be synthesized without errors. As for functional correctness I can confirm that *nv_small* passes sanity checks. Well, at least those checks I was able to run directly on my development board (including tests available in `verif` directory translated by myself into bare C). Not so sure about *master* - I think (it was quite a while ago) I had some problems with running reference DNN available in `sw` repository. But I didn't spent as much time for bringing up *master* branch as for *nv_small*.

Here are some utilization reports from my tests around various configurations in the `master` branch (synthesis only). Without any proof of functional correctness though.

[image]

The code base does not work with Vivado out-of-the-box. For example you need to set some global macros related to clock gating and use simplified RAM models for FPGA (you will find them in both branches). But in general there should be no reason for modifying Verilog code (well, I added some `'include` clauses here and there).

Further reading about NVDLA integration with FPGA (99% is about Zynq architecture) https://github.com/nvdla/hw/issues/110

xiaoguoer, 2019-04-07:
@mmaciag , hi. I just want to config registers manually(just like what your bare C code did), and I used NV_SMALL code. As for register address, I found /outdir/nv_small/spec/manual/opendla.h file. But I'm really confused about what's the real address of those registers? For example, the NVDLA_CDP.S_POINTER_0 in the [trace test of nv_small](https://github.com/nvdla/hw/blob/nv_small/verif/tests/trace_tests/nv_small/cdp_1x1x1_lrn3_int8_0/cdp_1x1x1_lrn3_int8_0.cfg). aforementioned opendla.h indicates its address is 0xd004, but I got 0 after I wrote 0xffff_ffff to this address. Could you give me some advice? Thank you !

mmaciag, 2019-04-08:
Hi @xiaoguoer. These addresses are relative to the base address of the entire NVDLA peripheral. The base address is out of scope of the NVDLA specification, because it depends on the physical mapping of the peripheral on the bus.

killerzula, 2019-09-19:
> But in general there should be no reason for modifying Verilog 

@mmaciag, what about [this](https://github.com/nvdla/hw/issues/306)?

nagendra7890, 2020-12-14:
Hi, @ghost I am stuck at the synthesis stage. If I consider the top module as NVDLA, showing modules files are not found. I tried to change the Verilog files as Global include. But still facing the same problem. If anyone solved this problem please help me to sort it out.
Please see the below-attached error file.

[image]

Thank You.