nvdla/hw #277: NVDLA Cmod library not found

luisarocas, 2019-03-26:
When following http://nvdla.org/vp.html, in step 2.4.1 I obtain the error: NVDLA Cmod library not found.

Everything has been followed as explained in nvdla.org/vp,  the system requirements are all met. I have tried to repeat the process several times and I keep receiving the same error.

The paths of vp and hw are well defined. Can someone help me?

[image]

prasshantg, 2019-03-27:
@luisarocas please share log for "tools/bin/tmake -build cmod_top"

luisarocas, 2019-03-27:
The log file outdir/build.log contains:

make: Entering directory `/home/luis/hw/spec/defs'
/home/utils/gcc-4.9.3/bin/cpp -undef -nostdinc -P -C nv_full.spec -o /home/luis/hw/outdir/nv_full/spec/defs/project.def
make: Leaving directory `/home/luis/hw/spec/defs'
make: Entering directory `/home/luis/hw/spec/manual'
/home/utils/java/jdk1.8.0_131/bin/java -jar Ordt.jar -parms test.parms -systemverilog /home/luis/hw/outdir/nv_full/spec/manual/sv/ -verilog /home/luis/hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /home/luis/hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /home/luis/hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /home/luis/hw/outdir/nv_full/spec/manual/dmod test.rdl
make: Leaving directory `/home/luis/hw/spec/manual'
make: Entering directory `/home/luis/hw/cmod'
mkdir -p ../outdir/nv_full/cmod/../cmod/nvdla_payload/
/home/utils/gcc-4.9.3/bin/g++ -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/usr/local/systemc-2.3.0//include -MMD -MP -fPIC -Wall -Werror -DSC_INCLUDE_DYNAMIC_PROCESSES -Wp,-w -std=c++11  -c ../cmod/nvdla_payload/dla_b_transport_payload.cpp -o ../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o
make: Leaving directory `/home/luis/hw/cmod'

fanqifei, 2019-03-27:
Hi @luisarocas, from outdir/build.log the cmod building looks like incomplete.
The sub-units like cdma, css are not built.
Could you please cleanup the cmod output directory and then perform a clean build?

luisarocas, 2019-03-27:
I notices that when doing sudo tools/bin/tmake -build cmod_top many errors appear. I still do not know what is wrong. Can someone help me?:

sudo tools/bin/tmake -build cmod_top
make: Entering directory '/home/luis/vp/hw/spec/defs'
/home/utils/gcc-4.9.3/bin/cpp -undef -nostdinc -P -C nv_full.spec -o /home/luis/vp/hw/outdir/nv_full/spec/defs/project.def
make: /home/utils/gcc-4.9.3/bin/cpp: Command not found
make: *** [/home/luis/vp/hw/outdir/nv_full/spec/defs/project.def] Error 127
Makefile:22: recipe for target '/home/luis/vp/hw/outdir/nv_full/spec/defs/project.def' failed
make: Leaving directory '/home/luis/vp/hw/spec/defs'
make: Entering directory '/home/luis/vp/hw/spec/manual'
/home/utils/java/jdk1.8.0_131/bin/java -jar Ordt.jar -parms test.parms -systemverilog /home/luis/vp/hw/outdir/nv_full/spec/manual/sv/ -verilog /home/luis/vp/hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /home/luis/vp/hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /home/luis/vp/hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /home/luis/vp/hw/outdir/nv_full/spec/manual/dmod test.rdl
make: /home/utils/java/jdk1.8.0_131/bin/java: Command not found
make: *** [default] Error 127
Makefile:20: recipe for target 'default' failed
make: Leaving directory '/home/luis/vp/hw/spec/manual'
make: Entering directory '/home/luis/vp/hw/cmod'
mkdir -p ../outdir/nv_full/cmod/../cmod/nvdla_payload/
/home/utils/gcc-4.9.3/bin/g++ -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/usr/local/systemc-2.3.0//include -MMD -MP -fPIC -Wall -Werror -DSC_INCLUDE_DYNAMIC_PROCESSES -Wp,-w -std=c++11  -c ../cmod/nvdla_payload/dla_b_transport_payload.cpp -o ../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o
make: /home/utils/gcc-4.9.3/bin/g++: Command not found
make: *** [../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o] Error 127
Makefile:165: recipe for target '../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o' failed
make: Leaving directory '/home/luis/vp/hw/cmod'
logfile: outdir/build.log
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.
luis@luis-Inspiron-5521:~/vp/hw$ sudo rm -rf outdir
luis@luis-Inspiron-5521:~/vp/hw$ sudo tools/bin/tmake -build cmod_top
make: Entering directory '/home/luis/vp/hw/spec/defs'
/home/utils/gcc-4.9.3/bin/cpp -undef -nostdinc -P -C nv_full.spec -o /home/luis/vp/hw/outdir/nv_full/spec/defs/project.def
make: /home/utils/gcc-4.9.3/bin/cpp: Command not found
make: *** [/home/luis/vp/hw/outdir/nv_full/spec/defs/project.def] Error 127
Makefile:22: recipe for target '/home/luis/vp/hw/outdir/nv_full/spec/defs/project.def' failed
make: Leaving directory '/home/luis/vp/hw/spec/defs'
make: Entering directory '/home/luis/vp/hw/spec/manual'
/home/utils/java/jdk1.8.0_131/bin/java -jar Ordt.jar -parms test.parms -systemverilog /home/luis/vp/hw/outdir/nv_full/spec/manual/sv/ -verilog /home/luis/vp/hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /home/luis/vp/hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /home/luis/vp/hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /home/luis/vp/hw/outdir/nv_full/spec/manual/dmod test.rdl
make: /home/utils/java/jdk1.8.0_131/bin/java: Command not found
make: *** [default] Error 127
Makefile:20: recipe for target 'default' failed
make: Leaving directory '/home/luis/vp/hw/spec/manual'
make: /home/utils/gcc-4.9.3/bin/g++: Command not found
make: *** [../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o] Error 127
make: Entering directory '/home/luis/vp/hw/cmod'
mkdir -p ../outdir/nv_full/cmod/../cmod/nvdla_payload/
/home/utils/gcc-4.9.3/bin/g++ -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/ [... cut ...]
Makefile:165: recipe for target '../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o' failed
make: Leaving directory '/home/luis/vp/hw/cmod'
logfile: outdir/build.log
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.22.1/IO/Tee.pm line 132.

mmaciag, 2019-03-27:
Isn't `sudo` messing something with paths?

prasshantg, 2019-03-29:
@luisarocas why are you using sudo? It will change the paths you have set.

javiergcanada, 2019-03-29:
Hello @luisarocas , I have recently experienced a similar problem to yours. (issue nvdla/sw#134 , please if someone knows how to correctly link it). I got the same error while trying to compile the CMOD library, but in my case its due to java, perl and the rest of the programs being located on /usr/bin, and the make instruction cannot obtain permission to deal with that folder.
I haven't found any solution yet, but make sure if that is nor your same issue

fanqifei, 2019-03-31:
Hi @luisarocas, did you follow http://nvdla.org/hw/v1/integration_guide.html or http://nvdla.org/vp.html to install the tools like g++, java?
The commands of running cpp and g++ are correct. The issue is that they were not found in /home/utils/gcc-4.9.3/bin. Could you please check whether they exist in that path? 
Using sudo is not necessary actually.

fanqifei, 2019-03-31:
Hi @javiergcanada , could you check the setting in tree.make? CPP, GCC are defined in tree.make.
Tree.make is generated in first step of http://nvdla.org/hw/v1/integration_guide.html#tree-build

javiergcanada, 2019-03-31:
Hi @fanqifei, as you can see in the screenshot i posted on nvdla/sw#134, I think I'm defining the location of CPP and GCC as /usr/bin correctly. I'm also not getting any error message on the next step, telling me that they were incorrectly located, it apparently just doesn't have permission to access those files. Thank you for your help.

mmaciag, 2019-04-01:
Maybe after executing `sudo` command now you have some leftovers belonging to `root`. Try something like:

```
find <your_hw_dir> -exec chown <your-user-name>:  {} \;
```
Or remove the project from disk and checkout from repository again.

As for configuring GCC location... Keep in mind that CMOD may still use gcc installed with the OS.  See https://github.com/nvdla/hw/pull/191

fanqifei, 2019-04-02:
> Hi @fanqifei, as you can see in the screenshot i posted on [nvdla/sw#134](https://github.com/nvdla/sw/issues/134), I think I'm defining the location of CPP and GCC as /usr/bin correctly. I'm also [...]

Hi @javiergcanada , in my tree.make, the setting are below:
CPP      := /home/utils/gcc-4.8.2/bin/cpp
GCC      := /home/utils/gcc-4.8.2/bin/gcc
When you run make to create tree.make, I think you should type the full path of cpp and gcc, like /home/utils/gcc-4.9.3/bin/cpp, not /usr/bin.

prasshantg, 2019-04-15:
@javiergcanada @luisarocas 

Please try with docker image (https://hub.docker.com/r/nvdla/vp) which has all dependencies install to build cmod and vp

Step 1. Run docker image
docker pull nvdla/vp
docker run -it -v /home:/home nvdla/vp

Step 2. Build CMOD
git clone https://github.com/nvdla/hw.git
cd hw/
git checkout origin/master
make

Creating tree.make to setup your working environment and projects
Enter project names (Press ENTER if use: nv_small nv_small_256 nv_small_256_full nv_medium_512 nv_medium_1024_full nv_large):nv_small
Using designware or not [1 for use/0 for not use] (Press ENTER if use: 1):
Enter design ware path (Press ENTER if use: /home/tools/synopsys/syn_2011.09/dw/sim_ver):
Enter c pre-processor path (Press ENTER if use: /home/utils/gcc-4.8.2/bin/cpp):/usr/bin/cpp
Enter gcc path (Press ENTER if use: /home/utils/gcc-4.8.2/bin/gcc):/usr/bin/gcc
Enter g++ path (Press ENTER if use: /home/utils/gcc-4.8.2/bin/g++):/usr/bin/g++
Enter perl path (Press ENTER if use: /home/utils/perl-5.10/5.10.0-threads-64/bin/perl):/usr/bin/perl
Enter java path (Press ENTER if use: /home/utils/java/jdk1.8.0_131/bin/java):/usr/bin/java
Enter systemc path (Press ENTER if use: /home/ip/shared/inf/SystemC/1.0/20151112/systemc-2.3.0/GCC472_64_DBG):/usr/local/systemc-2.3.0
Enter python path (Press ENTER if use: /home/tools/continuum/Anaconda3-5.0.1/bin/python):/usr/bin/python
Enter vcs_home path (Press ENTER if use: /home/tools/vcs/mx-2016.06-SP2-4):
Enter novas_home path (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
Enter verdi_home path (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
OPTIONAL: Enter verilator path (Press ENTER to use: verilator):
OPTIONAL: Enter clang path (Press ENTER to use: /home/utils/llvm-4.0.1/bin/clang):

tools/bin/tmake -build cmod_top

Step 3. Build VP

cd ..

git clone https://github.com/nvdla/vp.git
cd vp/
git submodule update --init --recursive

cmake -DCMAKE_INSTALL_PREFIX=build -DSYSTEMC_PREFIX=/usr/local/systemc-2.3.0/ -DNVDLA_HW_PREFIX=<Path_to_hw_repo> -DNVDLA_HW_PROJECT=nv_small

Aditi896, 2020-08-24:
All the steps involved seemed to run successfully for me. But it said CMOD Library not found. 

The hardware project is named nv_full by default. Files were generated under NVDLA/hw/outdir/nv_full/cmod. So when we run the cmake in the vp directory, give -DNVDLA_HW_PROJECT=nv_full in place of -DNVDLA_HW_PROJECT=nv_small