nvdla/hw #280: Why batch normalize is implemented independently in SDP?

keqian9, 2019-04-01:
Hi,
Does anyone known why batch normalize is implemented independently in SDP? it should be implemented with conv together.

Thanks.

silvaurus, 2019-04-04:
I think it is partly because in some networks BN are defined as a separate layer.
I don't think NVDLA's compiler will merge or partition layers defined in caffe model. So if the model defines BN as a separate layer, the compiler will assume so, then SDP will work in a separate mode, which means loading data from SDP_RDMA, instead of from CACC.

fanqifei, 2019-08-14:
Convolution and SDP can be fused in one HW layer in NVDLA.

Erdbeerenjanzi, 2019-09-04:
If you compile your network with the _fast-math_ flag, then BN is actually combined with the CONV layer. The _basic_ flag which is used as default seems to treat them independently.