nvdla/hw #281: nv_small AXI 128-bit configuration

flylcl, 2019-04-11:
Hi, 
Is there anyone who tested the AXI 128-bit in nv_small configuration?
There is a constraint in "projects.spec" file, said #error "one of NVDLA_PRIMARY_MEMIF_WIDTH_{256,128,64} must be set".
I just wonder it is runnable if I ignore this error constraint.
TY.

mmaciag, 2019-04-11:
What did you set in the nv_small.spec? PRIMARY_MEMIF_WIDTH_128 should work fine and project.spec should not detect it as an error. Are you working with `master` branch? There is nv_medium_512.spec and it is set to 128-bit.

flylcl, 2019-04-11:
Hi, mmaciag,
I am working on nv_small branch. And set the MEMIF_WIDTH to 128 as you said.
The error message is typo, it said said "one of NVDLA_PRIMARY_MEMIF_WIDTH_{512,64} must be set" in nv_small branch.
So, nv_small doesn't support nv_small configuration and I should exploit master branch?
TY.

mmaciag, 2019-04-11:
Nv_small is not configurable, that's why the branch is called nv_small - you may change the default nv_small.spec, however in most cases it will end up with synthesis errors. The `master` branch keeps what [roadmap](http://nvdla.org/roadmap.html#h1) calls *RTL support for fine grained configuration control*.

flylcl, 2019-04-15:
Hi, mmaciag,
Can I ask 1 more question?
There are several configuration specs in Master branch. So Can I modify the parameters in the file NVDLA provides or Can I create my own spec file with respect to constraints(restriction)?

mmaciag, 2019-04-16:
You can modify existing file or create your own. I don't know how well it is constrained - I was lucky enough to not see any synthesis errors and I tested various custom specs. 

Note that these spec files can give quite a lot of configurations. I don't believe that literally _all_ of them were tested an run by authors, so I wouldn't be surprised if certain configurations did not pass functional verification.

flylcl, 2019-04-16:
I got it. Thanks for your kind reply.

closed 2019-04-16 by flylcl