nvdla/hw #282: so many errors

long-a-zhang-tieto, 2019-04-11:
./tools/bin/tmake -build verilator

show errors:
V_NVDLA_MCIF_READ_eg.v.st:829:25: warning: missing terminating " character
 _&slsl&_        vprinti "
                         ^
warning: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Wunknown-warning-option]
1 warning generated.

/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla.o): relocation R_X86_64_32S against symbol `VNV_nvdla::__Vtable2_NV_nvdla__DOT__u_partition_o__DOT__u_NV_NVDLA_csb_master__DOT__u_fifo_csb2nvdla__DOT__wr_popping_gray_cntr' can not be used when making a PIE object; recompile with -fPIC

Hong-333, 2019-04-22:
refer to the answer in https://github.com/nvdla/hw/issues/146#issuecomment-404773479

Okaymaddy, 2019-06-05:
Have you guys tried implementing this on an FPGA?