nvdla/hw #283: error: ‘class scsim::cmod::CNVDLA_GEC_ERRSLICE0_MISSIONERR_ENABLE’ has no member named ‘uERR1’; did you mean ‘uERR0’?

piotr-aldec, 2019-04-12:
Hello,

I'm new with nvdla. I was able to build and run a simulation with Vivado with nvdlav1 branch and nv_full configuration. I was able also to run it on Riviera-PRO simulator as well.
However, this version is too big and I'd like to try nv_small_256.
This nv_small_256 version is available only in master branch and only vcs is supported.
I've called ./tools/bin/tmake -build ready_for_test
and got the following error:
[TMAKE]: building nv_small_256 in verif/vip/reference_model/nvdla_cmod_wrap 
Makefile:34: warning: overriding recipe for target '../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/Makefile:162: warning: ignoring old recipe for target '../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/glb/gen/gec_reg_model.cpp: In member function ‘void scsim::cmod::gec_reg_model::GecUpdateVariables(scsim::cmod::CNVDLA_GEC_REGSET*)’:
../../../../cmod/glb/gen/gec_reg_model.cpp:100:89: error: ‘class scsim::cmod::CNVDLA_GEC_ERRSLICE0_MISSIONERR_ENABLE’ has no member named ‘uERR1’; did you mean ‘uERR0’?
         errslice0_missionerr_enable_err1_ = reg_group_ptr->rERRSLICE0_MISSIONERR_ENABLE.uERR1();
                                                                                         ^~~~~
                                                                                         uERR0
make: *** [Makefile:39: ../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/glb/gen/gec_reg_model.ref.o] Error 1
[TMAKE]: DIE when building nv_small_256 verif/vip/reference_model/nvdla_cmod_wrap at ./tools/bin/tmake line 278.
[TMAKE]: nv_small_256: FAIL

What is a little bit strange for me is that I was not able to find uERR0 in any source files.

Could you please help with this issue?
                                                                                 
PROJECTS := nv_small_256

##=======================                                                                                 
##Linux Environment Setup                                                                                 
##=======================                                                                                 

USE_DESIGNWARE  := 0
DESIGNWARE_DIR  := 
CPP  := /usr/bin/cpp
GCC  := /usr/bin/g++
CXX  := /usr/bin/g++
PERL := /usr/bin/perl
JAVA := /usr/bin/java
SYSTEMC := /space_1/storage/SVN/test_base/branches/classes/lib_sim/systemc-2.3.1
PYTHON := /usr/bin/python3
VCS_HOME := 
NOVAS_HOME := 
VERDI_HOME := 
VERILATOR := verilator
CLANG := /usr/bin/clang

Thank you in advance,
Best Regards,
Piotr

fanqifei, 2019-08-14:
Hi @piotr-aldec , the class CNVDLA_GEC_REGSET is defined by macro MK_UREGSET_CLASS in umacros.uh. The root cause can't be found out at a first glance.
gec_reg_model.cpp is also compiled while building cmod. You can compare the compiler options first.

xyy-casia, 2020-04-22:
Hi@fanqifei , May I ask what specific action shall I take to fix this problem? Thanks a lot!