nvdla/hw #285: step by step build cmod_top error.

long-a-zhang-tieto, 2019-04-18:
zhanglong@onap-2:~/nvdla$ git clone https://github.com/nvdla/hw.git
Cloning into 'hw'...
remote: Enumerating objects: 7406, done.
remote: Total 7406 (delta 0), reused 0 (delta 0), pack-reused 7406
Receiving objects: 100% (7406/7406), 20.13 MiB | 2.28 MiB/s, done.
Resolving deltas: 100% (5089/5089), done.
zhanglong@onap-2:~/nvdla$ cd hw/
zhanglong@onap-2:~/nvdla/hw$ git checkout origin/master
Note: checking out 'origin/master'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 7cf6ad5 Add RAM model
zhanglong@onap-2:~/nvdla/hw$ make
Creating tree.make to setup your working environment and projects
Enter project names      (Press ENTER if use: nv_small nv_small_256 nv_small_256_full nv_medium_512 nv_medium_1024_full nv_large):nv_small
Using designware or not [1 for use/0 for not use] (Press ENTER if use: 1):
Enter design ware path (Press ENTER if use: /home/tools/synopsys/syn_2011.09/dw/sim_ver):
Enter c pre-processor path (Press ENTER if use: /home/utils/gcc-4.8.2/bin/cpp):/usr/bin/cpp
Enter gcc path             (Press ENTER if use: /home/utils/gcc-4.8.2/bin/gcc):/usr/bin/gcc
Enter g++ path             (Press ENTER if use: /home/utils/gcc-4.8.2/bin/g++):/usr/bin/g++
Enter perl path            (Press ENTER if use: /home/utils/perl-5.10/5.10.0-threads-64/bin/perl):/usr/bin/perl
Enter java path            (Press ENTER if use: /home/utils/java/jdk1.8.0_131/bin/java):/usr/bin/java
Enter systemc path         (Press ENTER if use: /home/ip/shared/inf/SystemC/1.0/20151112/systemc-2.3.0/GCC472_64_DBG):/usr/local/systemc-2.3.0
Enter python path          (Press ENTER if use: /home/tools/continuum/Anaconda3-5.0.1/bin/python):/usr/bin/python
Enter vcs_home path        (Press ENTER if use: /home/tools/vcs/mx-2016.06-SP2-4):
Enter novas_home path      (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
Enter verdi_home path      (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
OPTIONAL: Enter verilator path (Press ENTER to use: verilator):
OPTIONAL: Enter clang path     (Press ENTER to use: /home/utils/llvm-4.0.1/bin/clang):
=====================================================================
tree.make is created successfully, and you can edit tree.make manually if necessary
=====================================================================
zhanglong@onap-2:~/nvdla/hw$ tools/bin/tmake -build cmod_top
[TMAKE]: outdir does not exist, creating before build
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in cmod 
In file included from ../cmod/hls/cdma_libs/cdma_cvt.h:15:0,
                 from ../cmod/hls_wrapper/cdma_hls_wrapper.cpp:14:
../cmod/hls/include/ac_fixed.h: In instantiation of ‘ac_int<((I > 1) ? I : 1), S> ac_fixed<W2, I2, S2, Q2, O2>::to_ac_int() const [with int W = 26; int I = 26; bool S = true; ac_q_mode Q = (ac_q_mode)4; ac_o_mode O = (ac_o_mode)1]’:
../cmod/hls/include/nvdla_int.h:150:11:   required from ‘ac_int<oWidth, true> IntSignedShiftRight(ac_int<iWidth, true>, ac_int<sWidth, true>) [with unsigned int iWidth = 12; unsigned int sWidth = 6; unsigned int oWidth = 26]’
../cmod/hls/include/nvdla_float.h:1540:90:   required from ‘ac_int<iWidth, true> FpFloatToInt(ac_int<((1 + oExpoWidth) + MantWidth), true>) [with unsigned int IntWidth = 16; unsigned int ExpoWidth = 5; unsigned int MantWidth = 10]’
../cmod/hls/include/nvdla_float.h:1636:38:   required from here
../cmod/hls/include/ac_fixed.h:410:108: error: invalid operands of types ‘<unresolved overloaded function type>’ and ‘int’ to binary ‘operator<’
   inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).slc<AC_MAX(I,1)>(0); }
make: *** [../outdir/nv_small/cmod/../cmod/hls_wrapper/cdma_hls_wrapper.cpp.o] Error 1
[TMAKE]: DIE when building nv_small cmod at tools/bin/tmake line 278.

long-a-zhang-tieto, 2019-04-18:
zhanglong@onap-2:~/nvdla/hw$ make
Creating tree.make to setup your working environment and projects
Enter project names      (Press ENTER if use: nv_small nv_small_256 nv_small_256_full nv_medium_512 nv_medium_1024_full nv_large):nv_small
Using designware or not [1 for use/0 for not use] (Press ENTER if use: 1):
Enter design ware path (Press ENTER if use: /home/tools/synopsys/syn_2011.09/dw/sim_ver):
Enter c pre-processor path (Press ENTER if use: /home/utils/gcc-4.8.2/bin/cpp):/usr/bin/cpp
Enter gcc path             (Press ENTER if use: /home/utils/gcc-4.8.2/bin/gcc):/usr/bin/gcc
Enter g++ path             (Press ENTER if use: /home/utils/gcc-4.8.2/bin/g++):/usr/bin/g++
Enter perl path            (Press ENTER if use: /home/utils/perl-5.10/5.10.0-threads-64/bin/perl):/usr/bin/perl
Enter java path            (Press ENTER if use: /home/utils/java/jdk1.8.0_131/bin/java):/usr/bin/java
Enter systemc path         (Press ENTER if use: /home/ip/shared/inf/SystemC/1.0/20151112/systemc-2.3.0/GCC472_64_DBG):/usr/local/systemc-2.3.0
Enter python path          (Press ENTER if use: /home/tools/continuum/Anaconda3-5.0.1/bin/python):/usr/bin/python
Enter vcs_home path        (Press ENTER if use: /home/tools/vcs/mx-2016.06-SP2-4):
Enter novas_home path      (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
Enter verdi_home path      (Press ENTER if use: /home/tools/debussy/verdi3_2016.06-SP2-9):
OPTIONAL: Enter verilator path (Press ENTER to use: verilator):
OPTIONAL: Enter clang path     (Press ENTER to use: /home/utils/llvm-4.0.1/bin/clang):
\=====================================================================
tree.make is created successfully, and you can edit tree.make manually if necessary
\=====================================================================

=====================================================================
zhanglong@onap-2:~/nvdla/hw$ tools/bin/tmake -build cmod_top
[TMAKE]: outdir does not exist, creating before build
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in cmod 
In file included from ../cmod/hls/cdma_libs/cdma_cvt.h:15:0,
                 from ../cmod/hls_wrapper/cdma_hls_wrapper.cpp:14:
../cmod/hls/include/ac_fixed.h: In instantiation of ‘ac_int<((I > 1) ? I : 1), S> ac_fixed<W2, I2, S2, Q2, O2>::to_ac_int() const [with int W = 26; int I = 26; bool S = true; ac_q_mode Q = (ac_q_mode)4; ac_o_mode O = (ac_o_mode)1]’:
../cmod/hls/include/nvdla_int.h:150:11:   required from ‘ac_int<oWidth, true> IntSignedShiftRight(ac_int<iWidth, true>, ac_int<sWidth, true>) [with unsigned int iWidth = 12; unsigned int sWidth = 6; unsigned int oWidth = 26]’
../cmod/hls/include/nvdla_float.h:1540:90:   required from ‘ac_int<iWidth, true> FpFloatToInt(ac_int<((1 + oExpoWidth) + MantWidth), true>) [with unsigned int IntWidth = 16; unsigned int ExpoWidth = 5; unsigned int MantWidth = 10]’
../cmod/hls/include/nvdla_float.h:1636:38:   required from here
../cmod/hls/include/ac_fixed.h:410:108: error: invalid operands of types ‘<unresolved overloaded function type>’ and ‘int’ to binary ‘operator<’
   inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).slc<AC_MAX(I,1)>(0); }
make: *** [../outdir/nv_small/cmod/../cmod/hls_wrapper/cdma_hls_wrapper.cpp.o] Error 1
[TMAKE]: DIE when building nv_small cmod at tools/bin/tmake line 278.
[TMAKE]: nv_small: FAIL

long-a-zhang-tieto, 2019-04-18:
then
zhanglong@onap-2:~/nvdla/hw$ ./tools/bin/tmake -build vmod
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in vmod/vlibs 
[TMAKE]: building nv_small in vmod/include 
[TMAKE]: building nv_small in vmod/rams/model 
[TMAKE]: building nv_small in vmod/rams/synth 
[TMAKE]: building nv_small in vmod/rams/fpga/model 
[TMAKE]: building nv_small in vmod/fifos 
[TMAKE]: building nv_small in vmod/nvdla/apb2csb 
[TMAKE]: building nv_small in vmod/nvdla/cdma 
[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
[TMAKE]: building nv_small in vmod/nvdla/cmac 
[TMAKE]: building nv_small in vmod/nvdla/cacc 
[TMAKE]: building nv_small in vmod/nvdla/sdp 
[TMAKE]: building nv_small in vmod/nvdla/pdp 
[TMAKE]: building nv_small in vmod/nvdla/cfgrom 
[TMAKE]: building nv_small in vmod/nvdla/cdp 
[TMAKE]: building nv_small in vmod/nvdla/bdma 
[TMAKE]: building nv_small in vmod/nvdla/rubik 
[TMAKE]: building nv_small in vmod/nvdla/car 
[TMAKE]: building nv_small in vmod/nvdla/glb 
[TMAKE]: building nv_small in vmod/nvdla/csb_master 
[TMAKE]: building nv_small in vmod/nvdla/nocif 
[TMAKE]: building nv_small in vmod/nvdla/retiming 
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: Done nv_small
[TMAKE]: nv_small: PASS

prasshantg, 2019-04-18:
@long-a-zhang-tieto  Are you using docker image for this? You can run these build commands from docker bash so that it will use pre-installed tools.

long-a-zhang-tieto, 2019-04-18:
do you mean in the docker image git clone the hw folder?

prasshantg, 2019-04-18:
you can git clone from system bash but run build related commands (make, tools/bin/tmake -build cmod_top) from docker image bash

long-a-zhang-tieto, 2019-04-18:
well, thank you so much Prashant.

zhanglong@onap-2:~/nvdla$ docker run -it -v /home:/home nvdla/vp
root@8ac2168cec4f:/# cd /home/zhanglong/nvdla/hw/
root@8ac2168cec4f:/home/zhanglong/nvdla/hw# tools/bin/tmake -build cmod_top
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in cmod 
[TMAKE]: Done nv_small
[TMAKE]: nv_small: PASS
root@8ac2168cec4f:/home/zhanglong/nvdla/hw#

closed 2019-04-18 by long-a-zhang-tieto