nvdla/hw #288: Build Test Bench without vcs?

jyj-yj, 2019-05-13:
I built nv_small RTL with [this guide](http://nvdla.org/hw/v2/environment_setup_guide.html#build-test-bench). I got following error and it seems like there's no vcs in my test environment.
```
esa@hush34:~/jyj/nvdla_2/hw$ ./tools/bin/tmake -build vmod
[TMAKE]: outdir does not exist, creating before build
[TMAKE]: building nv_small in spec/defs
[TMAKE]: building nv_small in spec/manual
[TMAKE]: building nv_small in spec/odif
[TMAKE]: building nv_small in vmod/vlibs
[TMAKE]: building nv_small in vmod/include
[TMAKE]: building nv_small in vmod/rams/model
[TMAKE]: building nv_small in vmod/rams/synth
[TMAKE]: building nv_small in vmod/rams/fpga/model
[TMAKE]: building nv_small in vmod/fifos
[TMAKE]: building nv_small in vmod/nvdla/apb2csb
[TMAKE]: building nv_small in vmod/nvdla/cdma
[TMAKE]: building nv_small in vmod/nvdla/cbuf
[TMAKE]: building nv_small in vmod/nvdla/csc
[TMAKE]: building nv_small in vmod/nvdla/cmac
[TMAKE]: building nv_small in vmod/nvdla/cacc
[TMAKE]: building nv_small in vmod/nvdla/sdp
[TMAKE]: building nv_small in vmod/nvdla/pdp
[TMAKE]: building nv_small in vmod/nvdla/cfgrom
[TMAKE]: building nv_small in vmod/nvdla/cdp
[TMAKE]: building nv_small in vmod/nvdla/bdma
[TMAKE]: building nv_small in vmod/nvdla/rubik
[TMAKE]: building nv_small in vmod/nvdla/car
[TMAKE]: building nv_small in vmod/nvdla/glb
[TMAKE]: building nv_small in vmod/nvdla/csb_master
[TMAKE]: building nv_small in vmod/nvdla/nocif
[TMAKE]: building nv_small in vmod/nvdla/retiming
[TMAKE]: building nv_small in vmod/nvdla/top
[TMAKE]: Done nv_small
[TMAKE]: nv_small: PASS
esa@hush34:~/jyj/nvdla_2/hw$ ./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small in spec/defs
[TMAKE]: building nv_small in spec/manual
[TMAKE]: building nv_small in verif/testbench/trace_generator
make: /home/tools/vcs/mx-2016.06-SP2-4/bin/vcs: Command not found
make: *** [vcs_build] Error 127
[TMAKE]: DIE when building nv_small verif/testbench/trace_generator at ./tools/bin/tmake line 278.
[TMAKE]: nv_small: FAIL
esa@hush34:~/jyj/nvdla_2/hw$ 
```
Is there a way to test bench without vcs?
FYI here's my tree.make.
```
## ================================================================
## NVDLA Open Source Project
##
## Copyright(c) 2016 - 2017 NVIDIA Corporation.  Licensed under the
## NVDLA Open Hardware License; Check LICENSE which comes with
## this distribution for more information.
## ================================================================

##=======================                                                           
## Project Name Setup, multiple projects supported
##=======================                                                           
PROJECTS := nv_small
                                                                                                                                                                        ##=======================                                                           
##Linux Environment Setup                                                           
##=======================                                                           
                                                                                                                                                                        USE_DESIGNWARE  := 0
DESIGNWARE_DIR  := /home/tools/synopsys/syn_2011.09/dw/sim_ver
CPP  := /usr/bin/cpp
GCC  := /usr/bin/gcc
CXX  := /usr/bin/g++
PERL := /usr/bin/perl
JAVA := /usr/bin/java
SYSTEMC := /home/ip/shared/inf/SystemC/1.0/20151112/systemc-2.3.0/GCC472_64_DBG
PYTHON := /usr/bin/python3
VCS_HOME := /home/tools/vcs/mx-2016.06-SP2-4
NOVAS_HOME := /home/tools/debussy/verdi3_2016.06-SP2-9
VERDI_HOME := /home/tools/debussy/verdi3_2016.06-SP2-9
VERILATOR := verilator
CLANG := /home/utils/llvm-4.0.1/bin/clang
~
```

closed 2019-06-04 by jyj-yj

prasshantg, 2019-06-04:
@jyj-yj wondering if you were able to resolve the issues or close as there was no response?

jyj-yj, 2019-06-04:
@prasshantg no response :p
btw I figured out how to run test(not this issue, [this one](http://nvdla.org/sw/test_application.html#runtime-test-application)), so there's no need to build test bench I guess.
I hope you find what you are looking for.

prasshantg, 2019-06-04:
@jyj-yj sorry for the no response. I wanted to make sure you are unblocked. Btw we have released NVDLA compiler for nv_small and nv_large if you are trying those configurations.

https://github.com/nvdla/sw/releases/tag/v1.2.0

killerzula, 2019-09-10:
@jyj-yj  @prasshantg  the vcs in this case is the Synopsys VCS right? I dont have a license for it. Can I bypass it and build the "ready_for_test" using tmake?

jyj-yj, 2019-09-16:
@killerzula yes, it's Synopsys VCS.
> btw I figured out how to run test(not this issue, [this one](http://nvdla.org/sw/test_application.html#runtime-test-application)), so there's no need to build test bench I guess.
In my case, as mentioned, I didn't need to build 'ready_for_test' any more. I only used files from "tmake -build vmod" to do what I wanted. Maybe you should open your own github issue.