nvdla/hw #290: Try to build KMD using Petalinux

minhoha, 2019-05-20:
Hi. I'm try to build KMD using Petalinux by Xilinx.

I made a module like this.

**$ petalinux-create -t modules -name nvdla**

I got 

**/home/esa/petalinux/NVDLA_zynq/project-spec/meta-user/recipes-modules/nvdla**

Should I copy all source files or directories in sw/kwd ??

Which files should modified?

Docs I refer to are as follows:

[https://webthesis.biblio.polito.it/9033/1/tesi.pdf]

In page 74, there are some modifications. However, I cannot find does parts.

Okaymaddy, 2019-10-16:
Hey, did you figure this yet?

killerzula, 2019-12-01:
@minhoha Have you made any progress since then?

LeiWang1999, 2021-03-08:
any progress?

killerzula, 2021-03-08:
@LeiWang1999 idk about everyone else but i gave up on this long ago. May pick it up sometime again in the future but im not working on nvdla anymore.