nvdla/hw #292: Regarding implementation of NVDIA accelerator in an FPGA

Okaymaddy, 2019-06-05:
Hi,

I am working on implementing the NVDLA in a Xilinx FPGA. I am using the Vivado tool for the same. So far I know that I have to generate the custom IP's for the NVDLA and the AXI interconnect and create a wrapper of them. But, I am stuck on how to create the IP for the NVDLA block.  I am really new in this field and If someone can elaborate on the steps it will be of great help.

Thank you

Ahmed1Zaky, 2019-07-09:
Hey, have you managed to do it ? I am trying to do the same.

Okaymaddy, 2019-07-22:
Nope not yet.

killerzula, 2019-09-10:
Any update guys?

nagendra7890, 2020-12-14:
Hi, @Okaymaddy  Did you solve the issue?

I am also trying to do the same but getting errors at the synthesis stage. can you please help me to sort it out?

Please see the below-attached file.

Thank You. 

[image]