nvdla/hw #293: nv_small verilator not building (missing fles)

nganjehloo, 2019-07-11:
Hello. I am trying to build nv_small with verilator. I've gotten two errors. The first was RAMPDP_80X17_GL_M2_D2 which i found and added to the verilator file list in the verif folder/ However I cannot find the which is the new error /NV_NVDLA_SDP_MRDMA_EG_lat_fifo_8x65
Is nv_small and verilator currently supported?

nganjehloo, 2019-07-12:
I ended up using nvdlav1 as that seemed a bit more stable. Still had issues. A -Wno flag needs to be changed because of some changes in clang. Also a compiler flag also needs to be added because of linker issues. Once you are finished building, most tests will run and pass except googlenet and alexnet. The provided test expects a particular value at some point from the csb despite the hardware actually handling that operation differently than what the test expects. The hardware returns a 1 when a 0 is expected. Check for details #219.

After fixing that, the default nv_full design should work through verilator.

killerzula, 2019-09-09:
hey, when I run the following command
`TOT > ./tools/bin/tmake -build vmod`
I dont get the `[TMAKE]: DONE` output as described in the NVDLA environment setup guide. Instead, I get the following output,
`==============================================
files are generated under /home/ricksanchez/Documents/NVDLA/hw/outdir/nv_full/vmod/nvdla/top
==============================================
make: Leaving directory '/home/ricksanchez/Documents/NVDLA/hw/vmod/nvdla/top'
logfile: outdir/build.log
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.26.1/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.26.1/IO/Tee.pm line 132.
==============================================
Filehandle GEN1 opened only for input at /usr/local/share/perl/5.26.1/IO/Tee.pm line 132.
`
Would you kindly tell me if this output is fine or is there a problem?

dian-lun-lin, 2021-06-02:
I had the same issue and it turns out those missing modules are in the /fifos directory. I end up adding these lines in my veriator makefile and it works:
```
-I../../outdir/nv_small/vmod/rams/model
-I../../outdir/nv_small/vmod/fifos
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_NRDMA_cq_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_BRDMA_cq_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_MRDMA_cq_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_PDP_WDMA_DAT_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_PDP_RDMA_lat_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_PDP_RDMA_ro_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_DP_intpinfo_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_DP_data_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_DP_sumpd_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_RDMA_lat_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_NRDMA_lat_fifo_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_BRDMA_lat_fifo_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_SDP_MRDMA_EG_lat_fifo_lib.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_RDMA_ro_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDMA_WT_8ATMM_fifo.v
-v ../../outdir/nv_small/vmod/fifos/NV_NVDLA_CDP_WDMA_dat_fifo.v
```