nvdla/hw #298: Compilation error while building NVDLA CMOD

asgarpasha, 2019-08-06:
I followed "http://nvdla.org/vp.html" document to get the NVDLA Virtual Platform up and running but encountered a compilation error in section 2.3.4. while executing the command "tools/bin/tmake -build cmod_top".

Using Ubuntu 18.04 LTS, all the required directories for systemC, g++, perl were provided properly in the tree.make file in the previous step in section 2.3.4

Attaching a screenshot of the compilation error below
[image]

ScottThomasCWU, 2019-08-08:
Your compiler doesn't like that you have a templated member function that hasn't been called out with the template keyword.

You can fix this by changing the line to:
return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) this).**template** slc<AC_MAX(I,1)>(0);

sthomdev, 2019-08-08:
You're getting this error because your compiler like that a templated member function is being used without the template keyword.

You can easily fix the error by changing the line to the following:
return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).**template** slc<AC_MAX(I,1)>(0);

If you run into any other errors of this kind (you will) you can use the same fix.

mmaciag, 2019-08-12:
That issue would be fixed long time ago if comments in the pull requests were considered. Sadly this repository seems to have 'nobody-cares' status at NVIDIA.

fanqifei, 2019-08-15:
Hi @asgarpasha , version 3.9.0 of ac_types is integrated to branch master. @mmaciag found such issue before. see https://github.com/nvdla/hw/pull/191#issuecomment-411647611 . branch nv_small is not updated yet. If you are using branch nv_small, you can perform a cherry-pick.

killerzula, 2019-09-19:
Hey, @asgarpasha. Which branch are you compiling? Is it master or nvdlav1?