nvdla/hw #304: Unable to build vmod on nvdla on nv_full

killerzula, 2019-08-29:
Hi, I am trying to build the vmod folder in the nv_full in the hw-nvdla1 folder. However, the NVDLA Environment Setup Guide says that the output after the following command, when run from TOT,...
`./tools/bin/tmake -build vmod`
 ...should be...
`[TMAKE]: DONE`  
however the output that I am receiving is 
`==============================================
files are generated under /home/usman/Documents/Work/repos/hw-nvdlav1/outdir/nv_full/vmod/nvdla/top
==============================================
make: Leaving directory '/home/usman/Documents/Work/repos/hw-nvdlav1/vmod/nvdla/top'
logfile: outdir/build.log
==============================================
Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.
==============================================
Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.
`
[image]

That is not the complete log. Please help me resolve this issue.

killerzula, 2019-09-12:
Solved it. I was using nvdlav1 instead of nvdlav2(nv_small or master branch in GitHub repo).

closed 2019-09-12 by killerzula

nikitha-vasanth, 2023-07-31:
Can you elaborate a little on the solution you have mentioned above?

shazib-summar, 2023-07-31:
Tough luck my friend. its been years since ive done any work on HW dev. I've since changed careers and have no idea what i wrote in the comment above.

Wish I couldve been of more help. Good luck tho.

nikitha-vasanth, 2023-07-31:
Oh okay, no problem. Thanks tho!