nvdla/hw #306: Errors in nv_small SDP files (NV_NVDLA_SDP_HLS_*.v)

killerzula, 2019-09-18:
Hello, everyone. 

After successfully synthesizing nv_small in Vivado, I wanted to port it to Quartus. After defining the macros and starting compilation a few errors were generated (see below)

`Error (10161): Verilog HDL error at NV_NVDLA_SDP_HLS_Y_int_core.v(85): object "chn_in_prdy_0" is not declared. Verify the object name is correct. If the name is correct, declare the object.
`
`Error (10161): Verilog HDL error at NV_NVDLA_SDP_HLS_Y_int_core.v(86): object "chn_alu_op_prdy_0" is not declared. Verify the object name is correct. If the name is correct, declare the object.
`
`Error (10161): Verilog HDL error at NV_NVDLA_SDP_HLS_Y_int_core.v(87): object "chn_mul_op_prdy_0" is not declared. Verify the object name is correct. If the name is correct, declare the object.
`
`Error (10161): Verilog HDL error at NV_NVDLA_SDP_HLS_Y_int_core.v(88): object "chn_out_pvld_0" is not declared. Verify the object name is correct. If the name is correct, declare the object.
`

I naturally asked myself; why Vivado was not throwing the errors that Quartus was so eager to do so, therefore, I synthesized the code on Vivado again and opened the file `NV_NVDLA_SDP_HLS_Y_int_core.v` and there indeed are highlighted lines (see below)
[image]

I basically have a few questions that I hoped anyone here could answer,

- Why is Vivado synthesizing the nv_small instead of throwing errors regarding the lack of declarations of "chn_mul_op_prdy_0"?

- Why are these errors not thrown during compilation?

- Why does Quartus not synthesize the code successfully like Vivado but instead terminates compilation due to previously described errors?

- How do I resolve said errors in both Quartus and Vivado?

I have not checked all files yet, but, there are a few other files with errors present that are not thrown at compilation time. 
Any help regarding said issue would be appreciated.

Thank you.

fady-da, 2019-11-18:
I am having the same errors. did you find a solution?
Thanks

killerzula, 2019-11-18:
I take it that you are also trying to port NVDLA on Quartus. In that case, no. I did not find any such solution. However, I was able to synthesize it on Quartus by manually editing the files that raise such errors. Also there might be some `reg` declarations that need to be converted to `wire` and vice versa. The RTL should synthesize after that. Let me know if you need more correspondence.

P.S. Do not modify the code when synthesizing on Vivado.

fady-da, 2019-11-18:
Hi,
I am trying to compile the nv_small with cadence. and getting the same errors in the assign statements.
how did you edit it ?

Thanks for the help

killerzula, 2019-11-19:
> How did you edit it?

The editing varies with regards to the issue. In the picture that I posted it says,
> `'chn_out_pvld_0' is not declared`

If you observe the commented portion of the file, there is a foreach loop at line 66 that terminates at line 81. In this loop (at line 75) there is the declaration for the missing wire 

`wire chn_out_pvld_${i};`

From this point it should be clear what to do in order to resolve said error. Just declare the signal yourself. The rest of the editing is also pretty much similar, just need to define missing wires and/or change the types of a few signals.