nvdla/hw #313: vivado errors of zcu102

Allahfan, 2019-10-22:
hi，I've created block design.There are some errors when I generate output products.thank you for your answer.
Vivado Commands General Messages:
[Ipptcl 7-5] XIT evaluation error: expected floating-point number but got ""
[IP_Flow 19-1747] Failed to deliver file '/devdata/vivado2019/vivado/Downloads/Vivado/2018.3/data/ip/xilinx/smartconnect_v1_0/xit/update_contents.xit': ERROR: [Common 17-39] 'xit::source_ipfile' failed due to earlier errors.
[IP_Flow 19-167] Failed to deliver one or more file(s).
Block design General Messages:
[IP_Flow 19-3541] IP Elaboration error: Failed to generate IP 'smartconnect_0'. Failed to generate 'Elaborate BD' outputs: Failed to elaborate IP.
[BD 41-1031] Hdl Generation failed for the IP Integrator design /devdata/hu/project/learn2/learn2.srcs/sources_1/bd/design_1/design_1.bd
[IP_Flow 19-3899] Cannot get the environment domain name variable for the component vendor name. Setting the vendor name to 'user.org'.

powderluv, 2020-01-07:
did you resolve this ?

nagendra7890, 2020-12-14:
@Allahfan did you solve this problem?
I am also facing the same kind of problems.

buloruabutata, 2025-02-28:
If you use smartconnect ip, you can try to replace it with interconnect ip.