nvdla/hw #317: error in nv_small_256_full

mouhuajun, 2019-11-20:
Use the master version of SW and HW to compile the nv_small_256_full version of VP. I know that this version of the compiler cannot be compiled successfully, but nvdla runtime compiles successfully, and VP compiles successfully, but there are problems when loading the model:

libnvdla<1> failed to open dla device
libnvdla<1> Out of bounds DLA instance 0 requested.
(DLA_TEST) Error 0x00000004: runtime->load failed (in RuntimeTest.cpp, function loadLoadable(), line 353)
(DLA_TEST) Error 0x00000004: (propagating from RuntimeTest.cpp, function run(), line 443)
(DLA_TEST) Error 0x00000004: (propagating from main.cpp, function launchTest(), line 87)
[image]

mouhuajun, 2019-11-20:
There are also problems when loading opendla.ko. Loading opendla_1.ko is normal, but the KMD I compiled generated opendla.ko instead of opendla_1.ko
# insmod opendla.ko 

[   41.446675] opendla: loading out-of-tree module taints kernel.
[image]

MahmoudDagher, 2021-09-22:
Did you solve this issue and how ? 
Thanks a lot :)