nvdla/hw #319: some errors when using systemc

CristinaZN, 2019-11-21:
i want to test whether systems is installed successfully, and it reported errors below:
vp/systemc-2.3.0a/objdir$ g++ test_SystemC.cpp 
In file included from test_SystemC.cpp:3:0:
/usr/include/systemc.h:136:16: error: ‘std::gets’ has not been declared
     using std::gets;
i used ubuntu 18.04 and g++ 7.4.0

closed 2019-11-30 by CristinaZN

Hassan313, 2020-06-09:
@CristinaZN Hi, What was the solution to the problem?

Thank you.

Dueschen, 2021-01-11:
Hello, I've found that std:gets has been removed in C++11. Using old versions may be a solution to this problem.
Please refer to this: [(https://stackoverflow.com/questions/38352801/systemc-error-with-the-library)]
Changing the command `g++` to `g++ std=c++98` may help.