nvdla/hw #330: NV_Small Ubuntu 18.04.3 Compilation Error

Hassan313, 2020-05-06:
I am running the below command for nv_small :

./tools/bin/tmake -build cmod_top

I get the below error:

[image]

The OS is Ubuntu 18.04.3 LTS \n \l. 

Can anybody helps?

Thank you very much.

Po0ria, 2021-01-27:
I get a similar error after executing ./tools/bin/tmake -build cmod_top

Po0ria, 2021-01-28:
So my problem was that I was compiling the system with the latest gcc and after building this package I compiled the rest of them using gcc 4.8.5 . So I just made gcc 4.8.5 my default GCC(cpp, g++, and gcc) and I didn't have to change anything and it went through completely. I hope it helps you guys!

Hassan313, 2021-01-28:
> So my problem was that I was compiling the system with the latest gcc and after building this package I compiled the rest of them using gcc 4.8.5 . So I just made gcc 4.8.5 my default GCC(cpp, g++, [...]

Hi Pooria,

Thank you for posting the solution to the problem.