nvdla/hw #334: RTL does not build with Verilator 4.039

jameshanlon, 2020-07-11:
To reproduce (having run `make` and `./tools/bin/tmake -build vmod`):
```
cd verif/verilator
verilator --lint-only -f verilator.f
%Error-TIMESCALEMOD: ../../outdir/nv_full/vmod/nvdla/top/NV_nvdla.v:12:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     ../../outdir/nv_full/vmod/rams/model/RAMDP_128X11_GL_M2_E2.v:11:8: ... Location of module with timescale
   11 | module RAMDP_128X11_GL_M2_E2 (CLK_R, CLK_W, RE, WE
      |        ^~~~~~~~~~~~~~~~~~~~~
%Error-TIMESCALEMOD: ../../outdir/nv_full/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v:10:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     ../../outdir/nv_full/vmod/rams/model/RAMDP_128X11_GL_M2_E2.v:11:8: ... Location of module with timescale
   11 | module RAMDP_128X11_GL_M2_E2 (CLK_R, CLK_W, RE, WE
      |        ^~~~~~~~~~~~~~~~~~~~~
%Error-TIMESCALEMOD: ../../outdir/nv_full/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v:1873:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     ../../outdir/nv_full/vmod/rams/model/RAMDP_128X11_GL_M2_E2.v:11:8: ... Location of module with timescale
   11 | module RAMDP_128X11_GL_M2_E2 (CLK_R, CLK_W, RE, WE
      |        ^~~~~~~~~~~~~~~~~~~~~
...
```
Where:
```
$ ~/verilator/bin/verilator --version
Verilator 4.039 devel rev v4.032-451-gbb8899f3
```

wsnyder, 2020-07-15:
Please see https://github.com/verilator/verilator/issues/2461
Can someone confirm that it is fixed using Verilator git head?  Thanks.

wsnyder, 2020-07-15:
BTW the Verilog source code should really be using "localparam" not "parameter" under functions, as there's no IEEE-legal way to override a parameter used under a function.

farhanaslam-lm, 2020-11-18:
Hi, 
following error occurs while runing ./tools/bin/tmake -build verilator
Can someone help to resolve the error? Thanks

command: cd verif/verilator;make PROJECT=nv_small 
rm -f ../../outdir/nv_small/verilator/nvdla.cpp
cp nvdla.cpp ../../outdir/nv_small/verilator/nvdla.cpp
make -C ../../outdir/nv_small/verilator -f VNV_nvdla.mk CC=/usr/bin/clang CXX=/usr/bin/clang++ VM_PARALLEL_BUILDS=1 
make[1]: Entering directory '/home/farhanaslam/Desktop/NVDLA/hw_new/outdir/nv_small/verilator'
/usr/bin/clang++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o nvdla.o nvdla.cpp
g++    nvdla.o verilated.o VNV_nvdla__ALL.a    -o VNV_nvdla -lm -lstdc++ 
VNV_nvdla.mk:63: recipe for target 'VNV_nvdla' failed
make[1]: Leaving directory '/home/farhanaslam/Desktop/NVDLA/hw_new/outdir/nv_small/verilator'
Makefile:28: recipe for target '../../outdir/nv_small/verilator/VNV_nvdla' failed
warning: unknown warning option '-Wno-bool-operation'; did you mean '-Wno-bool-conversion'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Wunknown-warning-option]
2 warnings generated.
/usr/bin/ld: nvdla.o: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: verilated.o: relocation R_X86_64_32S against symbol `_ZN9Verilated9s_flushCbE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla.o): relocation R_X86_64_32S against symbol `_ZN9VNV_nvdla116__Vtable3_NV_nvdla__DOT__u_partition_o__DOT__u_NV_NVDLA_csb_master__DOT__u_fifo_csb2nvdla__DOT__wr_popping_gray_cntrE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_MCIF_READ_EG_lat_fifo.o): relocation R_X86_64_32S against symbol `_ZN40VNV_nvdla_NV_NVDLA_MCIF_READ_EG_lat_fifo28__Vtable1___PVT__rq_rd_countE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_SDP_RDMA_EG_ro.o): relocation R_X86_64_32S against symbol `_ZN33VNV_nvdla_NV_NVDLA_SDP_RDMA_EG_ro25__Vtable1___PVT__beat_cntE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla__Slow.o): relocation R_X86_64_32S against symbol `_ZN9VNV_nvdla80__Vtable1_NV_nvdla__DOT__u_partition_o__DOT__u_NV_NVDLA_cfgrom__DOT__reg_rd_dataE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_partition_m__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_nv_ram_rws_256x64__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_CDP_WDMA_dat_fifo_32x8__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_CDMA_dual_reg__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_CDMA_CVT_cell__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_nv_ram_rws_16x64__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_SDP_REG_dual__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_CDP_RDMA_ro_fifo_32x8__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_PDP_RDMA_ro_fifo_32x8__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_PDP_WDMA_DAT_fifo_32x8__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_MCIF_READ_EG_lat_fifo__Slow.o): relocation R_X86_64_32S against symbol `_ZN40VNV_nvdla_NV_NVDLA_MCIF_READ_EG_lat_fifo28__Vtable1___PVT__rq_rd_countE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_NV_NVDLA_MCIF_READ_IG_bpt__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
[... cut ...]
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla_ScanShareSel_JTAG_reg_ext_cg__pi9__Slow.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: VNV_nvdla__ALL.a(VNV_nvdla__Syms.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
make[1]: *** [VNV_nvdla] Error 1
make: *** [../../outdir/nv_small/verilator/VNV_nvdla] Error 2

wsnyder, 2020-11-18:
@farhanaslam-lm this is a different error, so it is more "polite" to make a different issue rather than use an old one. But anyhow.

>warning: unknown warning option '-Wno-bool-operation'; did you mean '-Wno-bool-conversion'? [-Wunknown-warning-option]

This means your verilator was ./configured and built on a system or g++ different than you are using. This won't go well.

>usr/bin/ld: nvdla.o: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC

This means your gcc had different options for different .o files.

Put together perhaps you're using the wrong g++ for your system?  Anyhow this is a general C compiler problem likely nothing under Verilator's control.

farhanaslam-lm, 2020-11-19:
@wsnyder Thank you ver muc, it helped a lot

Bliipie, 2021-03-18:
@farhanaslam-lm Hey I've got the same error, did you manage to solve it, and if so what did you change?