nvdla/hw #335: systemc header error

ConstantPark, 2020-09-03:
I have a question about this error.

soc@esoc-desktop:~/1.Workspace_1/constant/array_processor/hw$ ./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small_256 in spec/defs
[TMAKE]: building nv_small_256 in spec/manual
[TMAKE]: building nv_small_256 in verif/testbench/trace_generator
 9:15:02 (snpslmd) OUT: "VCSCompiler_Net" esoc@esoc-desktop  [snps_checkout_1599092102]
 9:15:06 (snpslmd) OUT: "VT_Testbench" esoc@esoc-desktop  [snps_checkout_1599092102]
 9:15:14 (snpslmd) IN: "VCSCompiler_Net" esoc@esoc-desktop  [snps_checkout_1599092102]
 9:15:14 (snpslmd) IN: "VT_Testbench" esoc@esoc-desktop  [snps_checkout_1599092102]
 9:15:14 (snpslmd) OUT: "VCSCompiler_Net" esoc@esoc-desktop  [snps_checkout_1599092114]
 9:15:14 (snpslmd) IN: "VCSCompiler_Net" esoc@esoc-desktop  [snps_checkout_1599092114]
[TMAKE]: building nv_small_256 in spec/odif
[TMAKE]: building nv_small_256 in vmod/vlibs
[TMAKE]: building nv_small_256 in vmod/include
[TMAKE]: building nv_small_256 in vmod/rams/model
[TMAKE]: building nv_small_256 in vmod/rams/synth
[TMAKE]: building nv_small_256 in vmod/rams/fpga/model
[TMAKE]: building nv_small_256 in vmod/fifos
[TMAKE]: building nv_small_256 in vmod/nvdla/apb2csb
[TMAKE]: building nv_small_256 in vmod/nvdla/cdma
[TMAKE]: building nv_small_256 in vmod/nvdla/cbuf
[TMAKE]: building nv_small_256 in vmod/nvdla/csc
[TMAKE]: building nv_small_256 in vmod/nvdla/cmac
[TMAKE]: building nv_small_256 in vmod/nvdla/cacc
[TMAKE]: building nv_small_256 in vmod/nvdla/sdp
[TMAKE]: building nv_small_256 in vmod/nvdla/pdp
[TMAKE]: building nv_small_256 in vmod/nvdla/cfgrom
[TMAKE]: building nv_small_256 in vmod/nvdla/cdp
[TMAKE]: building nv_small_256 in vmod/nvdla/bdma
[TMAKE]: building nv_small_256 in vmod/nvdla/rubik
[TMAKE]: building nv_small_256 in vmod/nvdla/car
[TMAKE]: building nv_small_256 in vmod/nvdla/glb
[TMAKE]: building nv_small_256 in vmod/nvdla/csb_master
[TMAKE]: building nv_small_256 in vmod/nvdla/nocif
[TMAKE]: building nv_small_256 in vmod/nvdla/retiming
[TMAKE]: building nv_small_256 in vmod/nvdla/top
[TMAKE]: building nv_small_256 in verif/vip/reference_model/nvdla_cmod_wrap
Makefile:34: warning: overriding recipe for target '../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/Makefile:162: warning: ignoring old recipe for target '../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
In file included from ../../../../cmod/nvdla_payload/dla_b_transport_payload.cpp:11:0:
../../../../cmod/nvdla_payload/dla_b_transport_payload.h:14:19: fatal error: systemc: No such file or directory
compilation terminated.
make: *** [../../../../outdir/nv_small_256/verif/vip/reference_model/nvdla_cmod_wrap/nvdla_payload/dla_b_transport_payload.ref.o] Error 1
[TMAKE]: DIE when building nv_small_256 verif/vip/reference_model/nvdla_cmod_wrap at ./tools/bin/tmake line 278.
[TMAKE]: nv_small_256: FAIL

I set the correct systemc address in the tree.make file.
But I don't know how to solve this problem.
I never touch any makefile.

Dueschen, 2021-03-04:
I also encountered this problem when building test bench. Can someone please help us?

yanyicheng, 2024-01-10:
Hi,have you solved this problem? I have also encoutered this problem