nvdla/hw #341: Regarding IP creation for NVDLA using VIvado

nagendra7890, 2020-12-14:
I am stuck at the synthesis stage. If I consider the top module as NVDLA, showing modules files are not found. I tried to change the Verilog files as Global include. But still facing the same problem. If anyone solved this problem please help me to sort it out.
Please see the below-attached error file.

Thank You.
[image]

nagendra7890, 2021-02-16:
hello, @qaziullah  I am also getting the same error. if  someone solve please help me to sort it out

Thank You.

qaziullah, 2021-02-16:
@nagendra7890 which branch are you using?
The way it worked for me was that I used the name of project same as the branch name. 
if you are using nv_small branch then try naming the project nv_small and then go on with ./tools/bin/make -build vmod.

Just name the project the same as the branch (nv_small or nv_full) and make sure that there is  no spell or case difference.

nagendra7890, 2021-02-17:
> @nagendra7890 which branch are you using?
> The way it worked for me was that I used the name of project same as the branch name.
> if you are using nv_small branch then try naming the project nv_small [...]

nagendra7890, 2021-02-17:
@qaziullah I changed to nv_small.
I am able to generate cmod code but not vmod.do we need to add a project.h file in nv_small?
[image]

closed 2021-02-17 by nagendra7890

qaziullah, 2021-02-17:
[image]

Have you installed all the tool required as shown in the figure.