nvdla/hw #342: No Rule to Make nv_Full_prj.spec

qaziullah, 2021-01-07:
make: Entering directory '/home/qazi/hw/spec/defs'
make: *** No rule to make target 'nv_Full_prj.spec', needed by '/home/qazi/hw/outdir/nv_Full_prj/spec/defs/project.def'.  Stop.

qaziullah, 2021-01-07:
I didn't understand how to change the environment variable as mentioned in in the integration module. 
What is VCS and what VSIM? 

I run ./tools/bin/tmake  without changing any macro and got an error. 
error details from the build log is attached below.

files are generated under /home/qazi/new_folder/hw/outdir/nv_full/vmod/nvdla/top
==============================================
make: Leaving directory '/home/qazi/new_folder/hw/vmod/nvdla/top'
make: Entering directory '/home/qazi/new_folder/hw/verif/sim'
DUTDIR = dut
VIPDIR = vip
Using cleartext directories
/home/tools/vcs/mx-2015.09-SP2-9-T0426/bin/vcs -f /home/qazi/new_folder/hw/verif/dut/dut.f  -Xddg=0x1 -o ./simv -cpp /home/utils/gcc-4.7.2/bin/g++  +nospecify +notimingchecks +define+VLIB_NO_UDP +warn=noTFIPC +warn=noTMR -full64 -y /home/qazi/new_folder/hw/verif/synth_tb -y /home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs +incdir+/home/qazi/new_folder/hw/verif/synth_tb +incdir+/home/qazi/new_folder/hw/verif/dut +incdir+/home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs +incdir+/home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/include +incdir+/home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs +incdir+..  +define+DESIGNWARE_NOEXIST +vcs+lic+wait -sverilog +libext+.v +libext+.sv -timescale=1ns/1ns -debug_all +define+NVTOOLS_SYNC2D_GENERIC_CELL +define+NO_PERFMON_HISTOGRAM +define+PRAND_OFF  +define+NO_DUMPS /home/qazi/new_folder/hw/verif/synth_tb/tb_top.v /home/qazi/new_folder/hw/verif/synth_tb/csb_master.v /home/qazi/new_folder/hw/verif/synth_tb/csb_master_seq.v /home/qazi/new_folder/hw/verif/synth_tb/axi_slave.v /home/qazi/new_folder/hw/verif/synth_tb/id_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/memory.v /home/qazi/new_folder/hw/verif/synth_tb/memresp_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/raddr_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/slave_mem_wrap.v /home/qazi/new_folder/hw/verif/synth_tb/waddr_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/wdata_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/wstrb_fifo.v /home/qazi/new_folder/hw/verif/synth_tb/clk_divider.v /home/qazi/new_folder/hw/verif/synth_tb/slave2mem_rd.v /home/qazi/new_folder/hw/verif/synth_tb/slave2mem_wr.v /home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW02_tree.v /home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW_lsd.v /home/qazi/new_folder/hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW_minmax.v -l ./simv.compile.log    -Mdir=./simvCsrc ; ./checkcompile.pl ./simv.compile.log ./simv

checkcompile : Compile FAILED (Warnings=0, Errors=0, ExecutableReady=0) 
The executable (./simv) has been removed.

tirumalnaidu, 2021-01-19:
If you have Synopsys VCS tool, go to tree.make file and change the the following variables to the paths of tool on your PC:
VCS_HOME := /home/tools/vcs/mx-2016.06-SP2-4
NOVAS_HOME := /home/tools/debussy/verdi3_2016.06-SP2-9
VERDI_HOME := /home/tools/debussy/verdi3_2016.06-SP2-9

vsim is the executable present in the VCS tool directory which runs the simulation.
To build the verilog modules, run this command: `./tools/bin/tmake -build vmod`. There is no need of Synopsys VCS for this.

qaziullah, 2021-01-20:
okays thanks, I have done that and have generated Verilog files.
But there is an issue while compiling the code in Quartus and I am getting some error saying that $RollPLI is a system function and cannot be synthesized. 

Actually, I am following a master thesis (https://lup.lub.lu.se/student-papers/search/publication/9007070) and trying to create IP for NVDLA core. it is was mentioned in the thesis page 45(28) table 3.1 (verilog Macros), that some macros listed in the table should be defined in the include file to get rid of some error. 
I didn't quite understand those macros, Could you kindly look into those.?

rookie0620, 2022-11-13:
can i ask a question about how  do you solve the error that $RollPLI is a system function and cannot be synthesized, there are plenty of  $RollPLI in files

qaziullah, 2022-11-26:
@rookie0620 you have to define the macros given in the thesis file file linked above.