nvdla/hw #349: nv_small tmake building ready_for_test error

shemadolev, 2021-05-14:
We're trying to build a simple nv_small project. We've passed the stages in [Build Tree](http://nvdla.org/hw/v2/environment_setup_guide.html#build-tree) till `./tools/bin/tmake -build ready_for_test` which generates the following:
[image]

This is our logfile in `outdir/nv_small/verif/testbench/trace_player/sysc.log`:
[sysc.log](https://github.com/nvdla/hw/files/6478245/sysc.log)
We've noticed 2 things:
1. The include path for SystemC is not as we supplied in the `tree.make`:
[image]
2. Seems like it's trying to run a sysc version of 2.3.0, and we can't understand why: The Systemc's version in the path we supplied is 2.3.1 and - when we're trying to change the `SYSC_VER` to 231 in the Makefile in `verif/testbench/trace_player` we're getting this error:
[image]

Any help will be appreciated.

LeiWang1999, 2021-05-23:
I think you can try this docker container .~

```
docker pull farhanaslam/nvdla 
```

shemadolev, 2021-05-30:
Solved:
1. We changed the version of systemc in : `hw/verif/testbench/traceplayer/Makefile`
2. We had a problem with the limit of file descriptors in our machine. Calling `limit descriptors 1024` solved the 'Bad file descriptor' error.

closed 2021-05-30 by shemadolev