nvdla/hw #351: Why sdp has two X module?

FengJungle, 2021-08-06:
[image]
As the picture shows,  the X1 and X2 are totally the same, why there is two X module in SDP?

FengJungle, 2021-08-06:
And one more question: Compared with X1/X2, the difference in Y is the LUT.  Why the SDP is X1 + X2 + Y(with LUT), but not a single X(with LUT)?  
Could anyone share the design idea? 
Thanks a lot!

bg193, 2021-09-19:
从设计上看可能是为了支持算子融合

FengJungle, 2021-09-19:
> 从设计上看可能是为了支持算子融合

你是说，计算图中可能有相邻的两个使用到sdp的算子可以融合到一个sdp完成吗？看nvdla的compiler源码的确有这样的操作

bg193, 2021-12-03:
> > 从设计上看可能是为了支持算子融合
> 
> 你是说，计算图中可能有相邻的两个使用到sdp的算子可以融合到一个sdp完成吗？看nvdla的compiler源码的确有这样的操作

理论上可以一个通路做bias，一个做norm

FengJungle, 2021-12-06:
> 

嗯，实际上源码里也的确有这样的融合的优化处理。不过一个x+y模块也可以完成算子融合，我只是很奇怪为什么是两个x而不是1个或者3个？文档里也没有给出说明。所以就想问问大家的看法。