nvdla/hw #368: Systemc error when building NVDLA Test Bench

yanyicheng, 2024-01-10:
Here is the log：

[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in vmod/vlibs 
[TMAKE]: building nv_small in vmod/include 
[TMAKE]: building nv_small in vmod/rams/model 
[TMAKE]: building nv_small in vmod/rams/synth 
[TMAKE]: building nv_small in vmod/rams/fpga/model 
[TMAKE]: building nv_small in vmod/fifos 
[TMAKE]: building nv_small in vmod/nvdla/apb2csb 
[TMAKE]: building nv_small in vmod/nvdla/cdma 
[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
[TMAKE]: building nv_small in vmod/nvdla/cmac 
[TMAKE]: building nv_small in vmod/nvdla/cacc 
[TMAKE]: building nv_small in vmod/nvdla/sdp 
[TMAKE]: building nv_small in vmod/nvdla/pdp 
[TMAKE]: building nv_small in vmod/nvdla/cfgrom 
[TMAKE]: building nv_small in vmod/nvdla/cdp 
[TMAKE]: building nv_small in vmod/nvdla/bdma 
[TMAKE]: building nv_small in vmod/nvdla/rubik 
[TMAKE]: building nv_small in vmod/nvdla/car 
[TMAKE]: building nv_small in vmod/nvdla/glb 
[TMAKE]: building nv_small in vmod/nvdla/csb_master 
[TMAKE]: building nv_small in vmod/nvdla/nocif 
[TMAKE]: building nv_small in vmod/nvdla/retiming 
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: building nv_small in verif/vip/reference_model/nvdla_cmod_wrap 
Makefile:34: warning: overriding recipe for target '../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/Makefile:162: warning: ignoring old recipe for target '../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
In file included from ../../../../cmod/nvdla_payload/dla_b_transport_payload.cpp:11:0:
../../../../cmod/nvdla_payload/dla_b_transport_payload.h:14:19: fatal error: systemc: No such file or directory
 #include <systemc>
                   ^
compilation terminated.
make: *** [../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/nvdla_payload/dla_b_transport_payload.ref.o] Error 1
[TMAKE]: DIE when building nv_small verif/vip/reference_model/nvdla_cmod_wrap at ./tools/bin/tmake line 278.
[TMAKE]: nv_small: FAIL

I have checked the file path of systemc, which is correct. Have anyone encoutered this kind of problem?

AlessandroVeronesi, 2024-04-09:
Can you share your `tree.make` file?
It may be that the actual SystemC path might be slightly different from what you specified in such file.

179691368, 2024-04-10:
##======================= 										  
## Project Name Setup, multiple projects supported			  	  
##======================= 										  
PROJECTS := nv_small
  																  
##======================= 										  
##Linux Environment Setup 										  
##======================= 										  
  																  
USE_DESIGNWARE      := 0
DESIGNWARE_DIR      := /home/tools/synopsys/syn_2011.09/dw/sim_ver
CPP      := /usr/bin/cpp
GCC      := /usr/bin/gcc
CXX      := /usr/bin/g++
PERL     := /usr/bin/perl
JAVA     := /usr/bin/java
SYSTEMC  := /usr/local/systemc-2.3.0
PYTHON   := /usr/bin/python3
VERDI_HOME := /usr/Synopsys/verdi/R-2020.12-SP1
NOVAS_HOME := /usr/Synopsys/verdi/R-2020.12-SP1
VCS_HOME   := /usr/Synopsys/vcs/Q-2020.03-SP2-7
CLANG      := /usr/bin/clang
VERILATOR  := verilator ////////////////Hello, I'm having the same issue. How did you solve it?

179691368, 2024-04-10:
> Can you share your `tree.make` file? It may be that the actual SystemC path might be slightly different from what you specified in such file.
Hello, I'm having the same issue, my tree.make is put below by me.

AlessandroVeronesi, 2024-04-10:
> ##=======================
>
> ## Project Name Setup, multiple projects supported
> ##======================= PROJECTS := nv_small
>
> ##======================= ##Linux Environment Setup [...]

"Unfortunately", the library path seems correct.
There are two major issues I experienced when generating the environment:

1. GCC version: for newer gcc versions, there are compatibility issues as reported alreeady in #330 and #298. A proposed bugfix have been posted in #191. Unfortunately, this solved the cmod compilation, but not the testbenches compilation in my case. In case this is your issue, I would recommend you to install a legacy GCC version which produces no problem (e.g., I am working with 4.8.5).

 2. Minor bug in the `tmake`'s `ready_for_test` target: it happened to me that if I run `tmake -build ready_for_test`, it ends with an error message if I previously generate part of the `outdir` content through other `tmake` targets. Since `ready_for_test` already generates the `vmod` and the `cmod` too, I would suggest to clean your `outdir` and directly run such target.

AlessandroVeronesi, 2024-04-10:
With respect to your comment about `verilator`, I cannot help since I only used Synopsys tools so far.
However, other people already reported issues when working with such tool (e.g., #334)